NVIDIA-ISAAC-ROS/isaac_ros_nvblox #5: Git clone is not working during 'Isaac ROS Docker Setup' part

beomsu7, 2022-04-01:
Hello Nvidia,

During [ROS2 Setup] - [Isaac ROS Docker Setup], the below command is not working well
>git clone --recurse-submodules https://gitlab-master.nvidia.com/isaac_ros/isaac_ros_nvblox.git  
  
  

The response is like this
>fatal: unable to access 'https://gitlab-master.nvidia.com/isaac_ros/isaac_ros_nvblox.git/': Could not resolve host: gitlab-master.nvidia.com

alexmillane, 2022-04-01:
Hi there,

Thanks for catching this. This line should read:

`git clone --recurse-submodules https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox.git`

We'll update the readme shortly.

beomsu7, 2022-04-01:
Thankyou for quick reply   

Acutally I tried with that command, and in that case, cloning the submodule part doesn't work correctly
The response is like below
>Cloning into 'isaac_ros_nvblox'...
remote: Enumerating objects: 108, done.
remote: Counting objects: 100% (108/108), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 108 (delta 13), reused 102 (delta 12), pack-reused 0
Receiving objects: 100% (108/108), 67.24 KiB | 11.21 MiB/s, done.
Resolving deltas: 100% (13/13), done.
Filtering content: 100% (7/7), 1.00 GiB | 27.51 MiB/s, done.
Submodule 'nvblox' (git@github.com:nvidia-isaac/nvblox.git) registered for path 'nvblox'
Cloning into '/workspaces/isaac_ros-dev/isaac_ws/src/isaac_ros_nvblox/nvblox'...
ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

>Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:nvidia-isaac/nvblox.git' into submodule path '/workspaces/isaac_ros-dev/isaac_ws/src/isaac_ros_nvblox/nvblox' failed
Failed to clone 'nvblox'. Retry scheduled
Cloning into '/workspaces/isaac_ros-dev/isaac_ws/src/isaac_ros_nvblox/nvblox'...
ssh: connect to host github.com port 22: Connection timed out
fatal: Could not read from remote repository.

>Please make sure you have the correct access rights
and the repository exists.
**fatal: clone of 'git@github.com:nvidia-isaac/nvblox.git' into submodule path '/workspaces/isaac_ros-dev/isaac_ws/src/isaac_ros_nvblox/nvblox' failed
Failed to clone 'nvblox' a second time, aborting**

AndreV84, 2022-04-03:
at another jetson
```
[ 58%] Built target test_mesh
[ 59%] Building CXX object tests/CMakeFiles/test_scene.dir/test_scene.cpp.o
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
tests/CMakeFiles/test_scene.dir/build.make:75: recipe for target 'tests/CMakeFiles/test_scene.dir/test_scene.cpp.o' failed
make[2]: *** [tests/CMakeFiles/test_scene.dir/test_scene.cpp.o] Error 4
CMakeFiles/Makefile2:883: recipe for target 'tests/CMakeFiles/test_scene.dir/all' failed
make[1]: *** [tests/CMakeFiles/test_scene.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```

AndreV84, 2022-04-03:
```
 git clone --recurse-submodules https://gitlab-master.nvidia.com/isaac_ros/isaac_ros_nvblox.git
Cloning into 'isaac_ros_nvblox'...
fatal: unable to access 'https://gitlab-master.nvidia.com/isaac_ros/isaac_ros_nvblox.git/': Could not resolve host: gitlab-master.nvidia.com
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws/src$ git clone --recurse-submodules https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox.git
Cloning into 'isaac_ros_nvblox'...
remote: Enumerating objects: 108, done.
remote: Counting objects: 100% (108/108), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 108 (delta 13), reused 102 (delta 12), pack-reused 0
Receiving objects: 100% (108/108), 67.24 KiB | 2.32 MiB/s, done.
Resolving deltas: 100% (13/13), done.
Submodule 'nvblox' (git@github.com:nvidia-isaac/nvblox.git) registered for path 'nvblox'
Cloning into '/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_nvblox/nvblox'...
fatal: unable to fork
fatal: clone of 'git@github.com:nvidia-isaac/nvblox.git' into submodule path '/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_nvblox/nvblox' failed
Failed to clone 'nvblox'. Retry scheduled
Cloning into '/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_nvblox/nvblox'...
fatal: unable to fork
fatal: clone of 'git@github.com:nvidia-isaac/nvblox.git' into submodule path '/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_nvblox/nvblox' failed
Failed to clone 'nvblox' a second time, aborting
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws/src$ 
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws/src$

AndreV84, 2022-04-03:
seems it wil require to run
```
git lfs pull

AndreV84, 2022-04-03:
then after manually cloning the nvblox it would build with stderr warning ``` 
[Processing: nvblox_ros]                                       
--- stderr: nvblox_ros                                        
CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_ros_lib".
This warning is for project developers.  Use -Wno-dev to suppress it.

qmake: could not find a Qt installation of ''
qmake: could not find a Qt installation of ''
qmake: could not find a Qt installation of ''
---
Finished <<< nvblox_ros [5min 7s]

Summary: 4 packages finished [5min 9s]
  1 package had stderr output: nvblox_ros

helenol, 2022-04-04:
@beomsu7 you can switch to https with this command from within the isaac_ros_nvblox repo:
```
git submodule set-url nvblox https://github.com/nvidia-isaac/nvblox.git
```

beomsu7, 2022-04-04:
@helenol  Thanks
It works well with docker and Isaac sim in ubuntu18.04 environment
Actually, first I used ubuntu20.04 and isaac sim, and there were few errors with ros2 bridge
I couldn't get ros2 topic from isaac sim in ubuntu20.04

Have you tried these with Isaac sim in ubuntu20.04?

helenol, 2022-04-04:
@beomsu7 on Jetson or x86? We're running 20.04 on x86 with Isaac sim for testing. Which version of Isaac Sim are you running?
Also for Isaac Sim with ROS2, **make sure your ROS2 workspace isn't sourced** in the window you start up our Isaac Sim script or the ROS2 bridge will crash Isaac Sim. :(

beomsu7, 2022-04-05:
@helenol on x86 and I tried isaac sim 21.2.1 and also in other verseions like 21.2.0, 21.1.1

And, mayb the 'make sure my ros2 workspace is sourced' means works with command 'source /opt/ros/foxy/setup.bash'
then, I did it.

My symptom is same with him (https://forums.developer.nvidia.com/t/standalone-ros2-bridge-fails-to-load/200201)
but, I gonna slowly try other stuff like change nvidia driver version or others

closed 2022-04-05 by beomsu7

alexmillane, 2022-04-05:
The comment is to make sure that your workspace **isn't** sourced. 

The reason is that Isaac Sim brings in its own version of ROS2 under the hood. If you source ROS2 before you run Isaac Sim, these two versions (your version and Isaac Sim's version) clash.

Can you start Isaac Sim through the GUI (omniverse launcher)?

beomsu7, 2022-04-05:
@alexmillane Ha ha... I was stupid..

After deleting ros2 source part from ~/.bashrc, I tried both launching isaac sim through the scripts and GUI

- GUI : 
    (1) run isaac sim
    (2) turn off ros bridge and turn on ros2 bridge from extension
    (3) add part(isaac examples - ros - navigation)
    (4) after loading, press 'play'
    (5) from docker, launch the carter_sim
    (6) the navigation is working but the 3d map is not displayed
[image]

- script
    (1) run the script(omni_python ~/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_nvblox/nvblox_isaac_sim/omniverse_scripts/carter_warehouse.py)
    (2) now there is an error like this

>>[ERROR]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to find shared library 'rmw_fastrtps_cpp', at /workspace/foxy_ws/src/rmw_implementation/rmw_implementation/src/functions.cpp:75', exiting with 1., at /workspace/foxy_ws/src/rcl/rcl/src/rcl/rmw_implementation_identifier_check.c:145

>>2022-04-05 08:11:46 [44,401ms] [Fatal] [carb.crashreporter-breakpad.plugin] libcarb.events.plugin.so!carbOnPluginStartup
2022-04-05 08:11:46 [44,403ms] [Fatal] [carb.crashreporter-breakpad.plugin] libcarb.tasking.plugin.so!std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (carb::tasking::Scheduler::*)(unsigned int, int, carb::cpp20::latch*), carb::tasking::Scheduler*, unsigned int, int, carb::cpp20::latch*> > >::_M_run()
2022-04-05 08:11:46 [44,406ms] [Fatal] [carb.crashreporter-breakpad.plugin] libpthread.so.0!funlockfile
2022-04-05 08:11:46 [44,409ms] [Fatal] [carb.crashreporter-breakpad.plugin] libtf.so!+0x3287f8
/home/ha/.local/share/ov/pkg/isaac_sim-2021.2.1/python.sh: line 43: 22501 Segmentation fault      (core dumped) $python_exe $@ $args

Whatever thanks very much

helenol, 2022-04-05:
1. Yes the issue is that we use different topic names in this case :/ You should be able to get this scenario to work by modifying the following in `carter_sim.launch.py`:
  a. /left/depth to /depth_left 
  b. /left/camera_info to /camera_info_left
  c. /left/color to /color_left
  d. /left/camera_info to /camera_info_left again
  e. Also in this scenario make sure you have a ROS clock that's set to sim time = true, but looks like that's already alright. Note that we changed the names in the script because the default names don't comply with ROS2 standards and therefore some visualization options don't work.
2. Looks like something ROS2 is sourced in your terminal there. Probably `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`. Try exporting `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` before running the OV script. With the OV script you don't need to change anything in the launch file above.