NVIDIA-ISAAC-ROS/isaac_ros_nvblox #6: third option for jetson

AndreV84, 2022-04-03:
you wrote
```
Therefore, to use nvblox on jetson you have two options:

1. manually compile ROS2 Foxy and required dependent packages from source
2. or use the Isaac ROS development Docker image from [Isaac ROS Common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common).
```
However, there is also a third option to do dist-upgrade on jetson that will result in 20.04 ubuntu system-wide
AV

AndreV84, 2022-04-03:
@hemalshahNV 
how to run nvbox with zed stereolabs zed camera connected to jetson?
how to run isaac_ros_nvblox with zed stereolabs camera?
is it possible to use both apps with the camera? any? neither of the two?

AndreV84, 2022-04-03:
```
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws$ rosdep install -i -r --from-paths src --rosdistro foxy -y --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv"
#All required rosdeps installed successfully
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws$ colcon build --packages-up-to nvblox_nav2 nvblox_ros nvblox_msgs nvblox_rviz_plugin
Starting >>> nvblox_msgs
Starting >>> isaac_ros_test
--- stderr: isaac_ros_test                                                          
/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< isaac_ros_test [3.01s]
Finished <<< nvblox_msgs [19.0s]                       
Starting >>> nvblox_nav2
Starting >>> nvblox_ros
Starting >>> nvblox_rviz_plugin
--- stderr: nvblox_ros                                                                                                           
CMake Error at CMakeLists.txt:62 (add_subdirectory):
  add_subdirectory given source "../nvblox/nvblox" which is not an existing
  directory.

CMake Error at CMakeLists.txt:104 (get_target_property):
  get_target_property() called with non-existent target "nvblox_interface".

CMake Error at CMakeLists.txt:111 (install):
  install TARGETS given target "nvblox_lib" which does not exist.

---
Failed   <<< nvblox_ros [8.33s, exited with code 1]
Aborted  <<< nvblox_nav2 [34.8s]                                                                                     
Aborted  <<< nvblox_rviz_plugin [45.5s]                                      

Summary: 2 packages finished [1min 5s]
  1 package failed: nvblox_ros
  2 packages aborted: nvblox_nav2 nvblox_rviz_plugin
  2 packages had stderr output: isaac_ros_test nvblox_ros
admin@nvidia-desktop:/workspaces/isaac_ros-dev/ros_ws$ 
```

helenol, 2022-04-04:
Jetpack 4.6.1 doesn't support Ubuntu 20.04. You need to either build ROS from source or build the docker image on 18.04. So no there's no 3rd option yet. Jetpack 5.0 will support 20.04 and then we'll have a nicer install process. :) I do not recommend doing a dist-upgrade.

You're missing rviz2. Either install rviz2 or use the docker image: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common

AndreV84, 2022-04-04:
@helenol 
Thank you for following up
```
You're missing rviz2. Either install rviz2 or use the docker image https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common
```
However, it is exactly the docker image I used, as it seems to me, as I used the docker steps for Xavier Jetson that clones the docker image https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common as one of steps

AndreV84, 2022-04-04:
here is the example how I run the isssac ros common container
```
 ~/workspaces/isaac_ros-dev/ros_ws/src/isaac_ros_common/scripts/run_dev.sh
isaac_ros_dev not specified, assuming /home/nvidia/workspaces/isaac_ros-dev
~ ~
Attaching to running container: isaac_ros_dev-aarch64-container
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ source /workspaces/isaac_ros-dev/ros_ws/install/setup.bash
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/camera/depth/camera_info -r depth/image:=/camera/depth/image_rect_raw 
I0404 16:04:06.438750 27193 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0404 16:04:06.439639 27193 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0404 16:04:06.439697 27193 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0404 16:04:06.439723 27193 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1649088246.448897243] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1649088246.449255690] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05

```

helenol, 2022-04-04:
@AndreV84 If you want to use the Zed camera in ROS2 with nvblox you will have to install the ROS2 Zed camera drivers. :) I don't, uh, really see a way around using the camera without running the drivers. 

You can run the library on Jetson without rviz (you can run rviz on another device, for example) with:
```
ros2 launch nvblox_nav2 carter_sim.launch.py run_rviz:=false
```

AndreV84, 2022-04-07:
@helenol now the situation has changed
I got zed ros2 node running
but nvblox got corrupted with the most recent changes as it seems to me
```
 source /workspaces/isaac_ros-dev/ros_ws/install/setup.bash
not found: "/workspaces/isaac_ros-dev/ros_ws/install/camera_info_manager/share/camera_info_manager/local_setup.bash"
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ source /workspaces/isaac_ros-dev/ros_ws/install/setup.bash
not found: "/workspaces/isaac_ros-dev/ros_ws/install/camera_info_manager/share/camera_info_manager/local_setup.bash"

AndreV84, 2022-04-07:
However, I can run the zed command you shared. but there are no vizualization outputs anywhere
```
 ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/camera/depth/camera_info -r depth/image:=/camera/depth/image_rect_raw  -p global_frame:=map
I0407 14:27:19.763958 29223 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0407 14:27:19.764844 29223 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0407 14:27:19.764941 29223 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0407 14:27:19.764971 29223 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1649341639.776463217] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1649341639.776912362] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05

AndreV84, 2022-04-07:
what image_common version is used in your container?
stereolabs folks pointed out that only https://github.com/ros-perception/image_common/releases/tag/3.0.0 will see zed topics

AndreV84, 2022-04-08:
seems we got ubuntu 20.04 for jetson in preview 5.0. already
so native method maybe will work better

AndreV84, 2022-04-14:
> ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/camera/depth/camera_info \
> -r depth/image:=/camera/depth/image_rect_raw -p global_frame:=

Hello @helenol 
I was able to figure out zed wraper setup on Jetpack 5.0; 
However the arguments you proposed above do not match nothing that zed camera provides as output topics.
Could you provide refined arguments for vizualization of zed outputs based onm the list below, please?
```
 ros2 topic list
/diagnostics
/parameter_events
/rosout
/tf
/tf_static
/zed/joint_states
/zed/robot_description
/zed/zed_node/confidence/confidence_map
/zed/zed_node/depth/camera_info
/zed/zed_node/depth/depth_info
/zed/zed_node/depth/depth_registered
/zed/zed_node/disparity/disparity_image
/zed/zed_node/left/camera_info
/zed/zed_node/left/image_rect_color
/zed/zed_node/left/image_rect_gray
/zed/zed_node/left_raw/camera_info
/zed/zed_node/left_raw/image_raw_color
/zed/zed_node/left_raw/image_raw_gray
/zed/zed_node/odom
/zed/zed_node/path_map
/zed/zed_node/path_odom
/zed/zed_node/point_cloud/cloud_registered
/zed/zed_node/pose
/zed/zed_node/pose_with_covariance
/zed/zed_node/rgb/camera_info
/zed/zed_node/rgb/image_rect_color
/zed/zed_node/rgb/image_rect_gray
/zed/zed_node/rgb_raw/camera_info
/zed/zed_node/rgb_raw/image_raw_color
/zed/zed_node/rgb_raw/image_raw_gray
/zed/zed_node/right/camera_info
/zed/zed_node/right/image_rect_color
/zed/zed_node/right/image_rect_gray
/zed/zed_node/right_raw/camera_info
/zed/zed_node/right_raw/image_raw_color
/zed/zed_node/right_raw/image_raw_gray
/zed/zed_node/stereo/image_rect_color
/zed/zed_node/stereo_raw/image_raw_color

AndreV84, 2022-04-14:
there is no vizualization output
```
ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -p global_frame:=map
I0414 03:08:54.979846 21379 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0414 03:08:54.980430 21379 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0414 03:08:54.980460 21379 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0414 03:08:54.980485 21379 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1649920134.988893534] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1649920134.989234345] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05
[INFO] [1649920135.088369260] [nvblox_node]: Depth frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1649920135.088571059] [nvblox_node]: Timing statistics: 

[INFO] [1649920145.137678686] [nvblox_node]: Depth frame statistics: 
avg=65.973510, min=11.000000, max=118.000000, std_dev=7.351125, count=151
[INFO] [1649920145.138326916] [nvblox_node]: Timing statistics: 
NVBlox Timing
-----------
esdf/integrate/compute/neighbors       	     62	00.016905	(00.000273 +- 00.000124)	[00.000151,00.000804]
esdf/integrate/compute/neighbors/kernel	     62	00.004362	(00.000070 +- 00.000018)	[00.000055,00.000171]
esdf/integrate/compute/neighbors/table 	     62	00.005375	(00.000087 +- 00.000063)	[00.000019,00.000436]
esdf/integrate/compute/sweep           	     62	00.004603	(00.000074 +- 00.000043)	[00.000029,00.000279]
esdf/integrate_slice                   	     19	00.085214	(00.004485 +- 00.004316)	[00.002121,00.021809]
esdf/integrate_slice/allocate          	     19	00.038929	(00.002049 +- 00.004119)	[00.000307,00.019020]
esdf/integrate_slice/clear             	     18	00.012554	(00.000697 +- 00.000131)	[00.000320,00.000928]
esdf/integrate_slice/compute           	     19	00.023626	(00.001243 +- 00.000338)	[00.000737,00.001914]
esdf/integrate_slice/mark_sites        	     19	00.009872	(00.000520 +- 00.000258)	[00.000261,00.001141]
in_view/combined_kernel                	     38	00.054325	(00.001430 +- 00.000033)	[00.001309,00.001502]
in_view/destroy                        	     38	00.000005	(00.000000 +- 00.000000)	[00.000000,00.000000]
in_view/output                         	     38	00.005345	(00.000141 +- 00.000056)	[00.000085,00.000371]
in_view/setup                          	     38	00.002612	(00.000069 +- 00.000085)	[00.000045,00.000583]
mesh/gpu/get_meshable                  	     38	00.022310	(00.000587 +- 00.000238)	[00.000290,00.001188]
mesh/gpu/integrate                     	     38	00.520089	(00.013687 +- 00.009910)	[00.009099,00.062586]
mesh/gpu/mesh_blocks                   	     38	00.480138	(00.012635 +- 00.009769)	[00.008249,00.061314]
mesh/gpu/mesh_blocks/block_allocation  	     38	00.086669	(00.002281 +- 00.005671)	[00.000241,00.035817]
mesh/gpu/mesh_blocks/copy_out          	     38	00.001687	(00.000044 +- 00.000050)	[00.000018,00.000248]
mesh/gpu/mesh_blocks/kernel_table      	     38	00.201008	(00.005290 +- 00.000396)	[00.004623,00.006128]
mesh/gpu/mesh_blocks/kernel_vertices   	     38	00.102247	(00.002691 +- 00.000229)	[00.002346,00.003478]
mesh/gpu/mesh_blocks/prep              	     38	00.087773	(00.002310 +- 00.005201)	[00.000357,00.021396]
ros/esdf                               	     19	00.086523	(00.004554 +- 00.004328)	[00.002172,00.021928]
ros/esdf/integrate                     	     19	00.086257	(00.004540 +- 00.004323)	[00.002160,00.021889]
ros/esdf/output                        	     19	00.000180	(00.000009 +- 00.000005)	[00.000007,00.000029]
ros/mesh                               	     38	00.571534	(00.015040 +- 00.011377)	[00.009760,00.072628]
ros/mesh/integrate_and_color           	     38	00.570881	(00.015023 +- 00.011374)	[00.009747,00.072598]
ros/mesh/output                        	     38	00.000235	(00.000006 +- 00.000001)	[00.000004,00.000008]
ros/total                              	    151	00.814180	(00.005392 +- 00.006805)	[00.000001,00.117208]
ros/tsdf                               	     76	00.153336	(00.002018 +- 00.001672)	[00.000001,00.021637]
ros/tsdf/conversions                   	     38	00.020501	(00.000539 +- 00.000113)	[00.000469,00.001197]
ros/tsdf/integrate                     	     38	00.130571	(00.003436 +- 00.002868)	[00.002349,00.020267]
ros/tsdf/transform                     	     38	00.001549	(00.000041 +- 00.000014)	[00.000030,00.000088]
tsdf/integrate                         	     38	00.122550	(00.003225 +- 00.002832)	[00.002216,00.019873]
tsdf/integrate/allocate_blocks         	     38	00.036769	(00.000968 +- 00.002560)	[00.000208,00.016155]
tsdf/integrate/get_blocks_in_view      	     38	00.062816	(00.001653 +- 00.000128)	[00.001521,00.002313]
tsdf/integrate/update_blocks           	     38	00.022201	(00.000584 +- 00.000209)	[00.000401,00.001382]

[INFO] [1649920155.147660883] [nvblox_node]: Depth frame statistics: 
avg=66.066445, min=11.000000, max=118.000000, std_dev=5.750248, count=301
[INFO] [1649920155.148317560] [nvblox_node]: Timing statistics: 
NVBlox Timing
-----------
[... cut ...]
ros/tsdf/transform                     	    150	00.005539	(00.000037 +- 00.000011)	[00.000029,00.000116]
tsdf/integrate                         	    150	00.405053	(00.002700 +- 00.000214)	[00.002112,00.019873]
tsdf/integrate/allocate_blocks         	    150	00.075725	(00.000505 +- 00.000131)	[00.000208,00.016155]
tsdf/integrate/get_blocks_in_view      	    150	00.244747	(00.001632 +- 00.000049)	[00.001479,00.002313]
tsdf/integrate/update_blocks           	    150	00.082245	(00.000548 +- 00.000080)	[00.000393,00.001382]

AndreV84, 2022-04-14:
seems it works 
but it doesn't display nothing
how do I get visdual outputs?
```
ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -p global_frame:=map

AndreV84, 2022-04-14:
@helenol 
[image]
is that it ? no coloring supported? am I missing something?

AndreV84, 2022-04-14:
Is it possible
to  in some way write(Save) one of current point clouds(meshes) ? in mesh/ point cloud file format?
@helenol 
also what is with nav2/ coloring? can not be used with rgb-d camera live input?

AndreV84, 2022-04-20:
@helenol 
Thank you for following up!
I can see the color information from terminal e.g. with
```
 ros2 topic hz /zed/zed_node/rgb/image_rect_color
average rate: 3.927
	min: 0.065s max: 0.610s std dev: 0.24505s window: 6
average rate: 7.115
	min: 0.055s max: 0.610s std dev: 0.16912s window: 18

```
but after I execute the following nvblox call, it doesn't seem to return color on /nvblox_node/mesh outputs, so it is grey only still
```
ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/color_info -p global_frame:=map

```
the problem probably is because this topic seems empty?

```
/zed/zed_node/rgb/color_info
```
 this doesn't seem to publish any data

were you able to vizualize colored mesh from rgbd camera at your setup with jetson or x86_64?

AndreV84, 2022-04-20:
probably you meant 

/zed/zed_node/rgb/camera_info
rather than 
/zed/zed_node/rgb/color_info

AndreV84, 2022-04-20:
but still no color
```
rosrun nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/camera_info -p global_frame:=map
I0420 01:08:13.289378  5872 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0420 01:08:13.290109  5872 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0420 01:08:13.290144  5872 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0420 01:08:13.290171  5872 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1650431293.300284067] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1650431293.300632295] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05
[INFO] [1650431408.944434788] [nvblox_node]: RGB frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650431408.952606139] [nvblox_node]: Depth frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650431408.952850686] [nvblox_node]: Timing statistics: 
NVBlox Timing
-----------
ros/color	      1	00.000003	(00.000003 +- 00.000000)	[00.000003,00.000003]
ros/total	      1	00.000015	(00.000015 +- 00.000000)	[00.000015,00.000015]

[INFO] [1650431409.349673370] [nvblox_node]: Got a new subscriber, sending entire map.
[ERROR] [1650431409.818310321] [nvblox_node]: Failed to transform color image.
[ERROR] [1650431410.231222328] [nvblox_node]: Failed to transform color image.
[ERROR] [1650431410.231421242] [nvblox_node]: Failed to transform color image.
[ERROR] [1650431411.506182894] [nvblox_node]: Failed to transform color image.
[ERROR] [1650431411.506396753] [nvblox_node]: Failed to transform color image.

AndreV84, 2022-04-20:
could you also share an example command for save_ply execution? How do I start the service from the terminal?
something like this won't work 
```
 ros2 run nvblox_ros nvblox_node --ros-args  -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/camera_info -p global_frame:=map -r save_ply output_dir=. 

```

AndreV84, 2022-04-20:
@helenol 
Thank you for following up
However execution of the command while generates some response both in local terminal and nvblox node, but it does  not seem to write file to anywhere. maybe output dir argument is missed?
```
ros2 service call /nvblox_node/save_ply std_srvs/Empty
requester: making request: std_srvs.srv.Empty_Request()

response:
std_srvs.srv.Empty_Response()

```

at nvblox node
```
W0420 08:50:41.036128  4493 ply_writer.cpp:26] Could not open file for PLY output.
W0420 08:50:41.343626  4493 ply_writer.cpp:26] Could not open file for PLY output.
W0420 08:50:41.916122  4493 ply_writer.cpp:26] Could not open file for PLY output.
[INFO] [1650459041.917257785] [nvblox_node]: Output PLY files to 
[INFO] [1650459046.164783201] [nvblox_node]: Depth frame statistics: 
avg=79.979920, min=0.000000, max=2340.000000, std_dev=155.531359, count=249
[INFO] [1650459046.165351842] [nvblox_node]: Timing statistics:

AndreV84, 2022-04-20:
@helenol 
seems I can write with the service call if using 
```
ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -p global_frame:=map -p output_dir:=.

```
Thanks
But still how to get color with zed camera? have you tested with any stereolabs or intel camera?

AndreV84, 2022-04-20:
from stereolabs:
```
What is the TF tree required by the application that you are performing? Does it follow the [REP105](https://www.ros.org/reps/rep-0105.html) specifications?
The ZED ROS wrapper follows the REP105 specifications, so it can be integrated into every application that correctly supports the ROS standard.
```

AndreV84, 2022-04-20:
@helenol what are my choices for getting the color?

AndreV84, 2022-04-20:
I have zed camera,It has two topics for color - one info and one color
-r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/camera_info 
they both kind of work as if I list their topics it will be shown in text terminal
but how would I resolve the error from nvblox_node? regarding the TF?

helenol, 2022-04-20:
You need to ensure that whatever frame_id the /zed/zed_node/rgb/image_rect_color topic has (header/frame_id) can be transformed into the global frame you have (`map` in your case) through the TF tree. Here is some docs on the TF tree and what its purpose is: https://docs.ros.org/en/foxy/Tutorials/Tf2/Introduction-To-Tf2.html 
This should be supplied by the driver. I don't know what's going wrong in your case. If it's available in the TF tree, as it is with the depth camera, then you should be getting the color images fused in. Since it's not for some reason (at least according to the error messages in your output), nvblox has no idea what to do with the color images since it doesn't know where they are in space and can't fuse them in.

AndreV84, 2022-04-21:
@helenol Thank you for the update
I passed your feedback to stereolabs. Maybe they will add on troubleshooting
I tried verifying tf using this command but it is probably errorneous
```
 ros2 run tf2_ros tf2_echo map /zed/zed_node/rgb/image_rect_color map
Failed to convert rate argument 'map' to a floating-point number

Myzhar, 2022-04-21:
Hi, I'm the Stereolabs developer working on the ZED ROS Wrapper.
I'm curious about this error message: `in tf2 frame_ids cannot start with a '/'`
Is it possible to get the TF tree to debug it?

Myzhar, 2022-04-21:
@AndreV84 you are passing the wrong parameters:
`/zed/zed_node/rgb/image_rect_color` is not a frame id, it's a topic name.
You must pass the correct frame ids where required.

AndreV84, 2022-04-21:
@Myzhar
which are the correct  frame ids of zed camera to pass as argument?

AndreV84, 2022-04-21:
@Myzhar 
```
ros2 run tf2_tools view_frames.py
[INFO] [1650535741.439487194] [view_frames]: Listening to tf data during 5 seconds...
[INFO] [1650535746.455180303] [view_frames]: Generating graph in frames.pdf file...
[INFO] [1650535746.460357531] [view_frames]: Result:tf2_msgs.srv.FrameGraph_Response(frame_yaml="base_link: \n  parent: 'odom'\n  broadcaster: 'default_authority'\n  rate: 15.196\n  most_recent_transform: 1650535746.389833\n  oldest_transform: 1650535741.322851\n  buffer_length: 5.067\nodom: \n  parent: 'map'\n  broadcaster: 'default_authority'\n  rate: 15.196\n  most_recent_transform: 1650535746.389833\n  oldest_transform: 1650535741.322851\n  buffer_length: 5.067\nzed_base_link: \n  parent: 'base_link'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_camera_center: \n  parent: 'zed_base_link'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_left_camera_frame: \n  parent: 'zed_camera_center'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_left_camera_optical_frame: \n  parent: 'zed_left_camera_frame'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_right_camera_frame: \n  parent: 'zed_camera_center'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_right_camera_optical_frame: \n  parent: 'zed_right_camera_frame'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\n")
```
[frames.pdf](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/files/8529990/frames.pdf)

```

Myzhar, 2022-04-21:
> @Myzhar which are the correct frame ids of zed camera to pass as argument?

They are in the header of each topic and you can read them in the PDF that you attached above

AndreV84, 2022-04-21:
so what causes the error and how to get the correct color?

Myzhar, 2022-04-21:
For example, this command
`ros2 run tf2_ros tf2_echo map /zed/zed_node/rgb/image_rect_color`
must be instead
`ros2 run tf2_ros tf2_echo map zed_left_camera_frame`

AndreV84, 2022-04-21:
```
 ros2 run tf2_ros tf2_echo map zed_left_camera_frame
[INFO] [1650536179.712090187] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650536181.675397085] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
At time 1650536183.555873048
- Translation: [-0.001, 0.060, 0.015]
- Rotation: in Quaternion [0.000, 0.025, -0.000, 1.000]
At time 1650536184.555792048
- Translation: [-0.001, 0.060, 0.015]
- Rotation: in Quaternion [0.000, 0.025, -0.000, 1.000]
At time 1650536185.555836048
- Translation: [-0.001, 0.060, 0.015]
- Rotation: in Quaternion [0.000, 0.025, -0.000, 1.000]
At time 1650536186.555901048
- Translation: [-0.001, 0.060, 0.015]
- Rotation: in Quaternion [0.000, 0.025, -0.000, 1.000]
^C[INFO] [1650536186.915386711] [rclcpp]: signal_handler(signal_value=2)
```
@Myzhar @helenol 
any ideas?

AndreV84, 2022-04-21:
why there is no color?

AndreV84, 2022-04-21:
```
 ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/camera_info -p global_frame:=map
I0421 06:27:14.357000  8796 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 06:27:14.357630  8796 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 06:27:14.357662  8796 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 06:27:14.357715  8796 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1650536834.365698824] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1650536834.366014970] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05
[INFO] [1650536836.765444720] [nvblox_node]: Depth frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650536836.765641563] [nvblox_node]: Timing statistics: 

[INFO] [1650536836.797969047] [nvblox_node]: RGB frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650536837.231978317] [nvblox_node]: Got a new subscriber, sending entire map.
[ERROR] [1650536838.004807942] [nvblox_node]: Failed to transform color image.
[ERROR] [1650536838.855563465] [nvblox_node]: Failed to transform color image.
[ERROR] [1650536838.855782997] [nvblox_node]: Failed to transform color image.
```
what I am getting is this

AndreV84, 2022-04-21:
is it a failure of nvblox core or of zed driver?
Or it can be resolved by adjusting the command that I run from local terminal?
Any ideas?
@Myzhar @helenol

Myzhar, 2022-04-21:
What's the output of the ZED Wrapper node?
I do not know how the `nvblox` node works, but I suppose that you must set the `publish_tf` parameter of the ZED node to false to not have more than a node publishing the `map` -> `odom` -> `base_link` transform

AndreV84, 2022-04-21:
@Myzhar 
The output of the zed node is as follows
```
ros2 launch zed_wrapper zed.launch.py
[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2022-04-21-06-08-48-200495-agx-8206
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [8209]
[INFO] [zed_wrapper-2]: process started with pid [8211]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed_base_link had 1 children
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1650535729.107045564 [zed.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1650535729.107236101 [zed.zed_state_publisher] [INFO] got segment zed_base_link
[robot_state_publisher-1] 1650535729.107273319 [zed.zed_state_publisher] [INFO] got segment zed_camera_center
[robot_state_publisher-1] 1650535729.107303401 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_frame
[robot_state_publisher-1] 1650535729.107331082 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_optical_frame
[robot_state_publisher-1] 1650535729.107364908 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_frame
[robot_state_publisher-1] 1650535729.107396846 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_optical_frame
[zed_wrapper-2] 1650535729.121290522 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650535729.121576201 [zed.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1650535729.121615019 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650535729.121646861 [zed.zed_node] [INFO]  * namespace: /zed
[zed_wrapper-2] 1650535729.121689647 [zed.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1650535729.121750770 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650535729.121855607 [zed.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1650535729.121986046 [zed.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-2] 1650535729.122040865 [zed.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1650535729.122136998 [zed.zed_node] [INFO]  * Camera model: zed - ZED
[zed_wrapper-2] 1650535729.122211850 [zed.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1650535729.122286446 [zed.zed_node] [INFO]  * SVO: live
[zed_wrapper-2] 1650535729.122358449 [zed.zed_node] [INFO]  * SVO Loop: FALSE
[zed_wrapper-2] 1650535729.122476183 [zed.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1650535729.122597182 [zed.zed_node] [INFO]  * Camera name: zed
[zed_wrapper-2] 1650535729.122699459 [zed.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1650535729.122807496 [zed.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1650535729.122884012 [zed.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1650535729.122963696 [zed.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1650535729.123065878 [zed.zed_node] [INFO]  * Camera framerate: 15
[zed_wrapper-2] 1650535729.123184636 [zed.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1650535729.123282177 [zed.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1650535729.123400743 [zed.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1650535729.123511853 [zed.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1650535729.123646580 [zed.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-2] 1650535729.123691766 [zed.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1650535729.123766170 [zed.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-2] 1650535729.123852926 [zed.zed_node] [INFO]  * [DYN] Image downsample factor: 1 
[zed_wrapper-2] 1650535729.123922626 [zed.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-2] 1650535729.124109452 [zed.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-2] 1650535729.124201904 [zed.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-2] 1650535729.124309782 [zed.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-2] 1650535729.124380474 [zed.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-2] 1650535729.124494751 [zed.zed_node] [INFO]  * [DYN] Gamma: 8
[zed_wrapper-2] 1650535729.124605829 [zed.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[zed_wrapper-2] 1650535729.124790095 [zed.zed_node] [INFO]  * [DYN] Exposure: 80
[zed_wrapper-2] 1650535729.124931094 [zed.zed_node] [INFO]  * [DYN] Gain: 80
[zed_wrapper-2] 1650535729.125027227 [zed.zed_node] [INFO]  * [DYN] Auto White Balance: TRUE
[zed_wrapper-2] 1650535729.125130112 [zed.zed_node] [INFO]  * [DYN] White Balance Temperature: 42
[zed_wrapper-2] 1650535729.125210596 [zed.zed_node] [INFO]  * Video QoS History: KEEP_LAST
[zed_wrapper-2] 1650535729.125308105 [zed.zed_node] [INFO]  * Video QoS History depth: 1
[zed_wrapper-2] 1650535729.125392174 [zed.zed_node] [INFO]  * Video QoS Reliability: RELIABLE
[zed_wrapper-2] 1650535729.125465649 [zed.zed_node] [INFO]  * Video QoS Durability: VOLATILE
[zed_wrapper-2] 1650535729.125498579 [zed.zed_node] [INFO] *** DEPTH parameters ***
[... cut ...]
[zed_wrapper-2] 1650535730.934499011 [zed.zed_node] [INFO]  * Translation: {0.001,0.000,-0.015}
[zed_wrapper-2] 1650535730.934536837 [zed.zed_node] [INFO]  * Rotation: {0.000,-2.865,0.000}
[zed_wrapper-2] 1650535731.034702100 [zed.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1650535731.034881853 [zed.zed_node] [INFO]  * T: [-1.66741e-19,0.06,0.015]
[zed_wrapper-2] 1650535731.034993283 [zed.zed_node] [INFO]  * Q: [0,0.0249967,0,0.999687]
[zed_wrapper-2] 1650535731.225245304 [zed.zed_node] [INFO] Odometry aligned to last tracking pose

Myzhar, 2022-04-21:
@AndreV84 as you can see in the ZED Wrapper console log you have all the information that you need:
TF frames, name of the published topics, ecc
The node is working as expected and all the TF are published.

AndreV84, 2022-04-21:
@Myzhar What exactly do you mean by saying to set zed node parameter to false? do you mean that I have to execute zed ros node with custom parameters? which exactly parameters? Could you specify please? I could interpret your message something like 
```
ros2 launch zed_wrapper zed.launch.py --ros-args -r publish_tf=false
``` 
but it is not a woprking command; what exactly i sthe command you suggest to execute to set the parameter to false?

AndreV84, 2022-04-21:
@helenol stereolab folks point out their driver works as expected
maybe it is an issue in nvblox code side?
what else?
AV

Myzhar, 2022-04-21:
`-r publish_tf=false`
this does not work. You must change the `common.yaml` file:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/config/common.yaml#L60

AndreV84, 2022-04-21:
so I just add it at the end of ros parameters in the yaml file like that? 
```
/**:
    ros__parameters:
        general:
            svo_file:                   ''
            svo_loop:                   false                               # Enable loop mode when using an SVO as input source
            svo_realtime:               false                               # if true SVO will be played trying to respect the original framerate eventually skipp>
            camera_timeout_sec:         5
            camera_max_reconnect:       5
            camera_flip:                false
            zed_id:                     0
            serial_number:              0
            resolution:                 2                                   # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
            sdk_verbose:                true
            grab_frame_rate:            15                                  # ZED SDK internal grabbing rate
            pub_frame_rate:             15.0                                # [DYNAMIC] - frequency of publishing of visual images and depth images
            gpu_id:                     -1
            publish_tf:                 false
```
how then I make sure the changes got applied?

AndreV84, 2022-04-21:
has it got applied as per the attached output?
```
ros2 launch zed_wrapper zed.launch.py
[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2022-04-21-09-25-54-156369-agx-10056
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [10059]
[INFO] [zed_wrapper-2]: process started with pid [10061]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed_base_link had 1 children
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1650547554.785883122 [zed.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1650547554.786100990 [zed.zed_state_publisher] [INFO] got segment zed_base_link
[robot_state_publisher-1] 1650547554.786143552 [zed.zed_state_publisher] [INFO] got segment zed_camera_center
[robot_state_publisher-1] 1650547554.786187042 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_frame
[robot_state_publisher-1] 1650547554.786238309 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_optical_frame
[robot_state_publisher-1] 1650547554.786266023 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_frame
[robot_state_publisher-1] 1650547554.786317769 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_optical_frame
[zed_wrapper-2] 1650547554.881028724 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650547554.881671863 [zed.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1650547554.881713977 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650547554.881766140 [zed.zed_node] [INFO]  * namespace: /zed
[zed_wrapper-2] 1650547554.881794942 [zed.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1650547554.881820351 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1650547554.881891523 [zed.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1650547554.882007657 [zed.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-2] 1650547554.882055660 [zed.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1650547554.882149649 [zed.zed_node] [INFO]  * Camera model: zed - ZED
[zed_wrapper-2] 1650547554.882243606 [zed.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1650547554.882361532 [zed.zed_node] [INFO]  * SVO: live
[zed_wrapper-2] 1650547554.882519429 [zed.zed_node] [INFO]  * SVO Loop: FALSE
[zed_wrapper-2] 1650547554.882629547 [zed.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1650547554.882754033 [zed.zed_node] [INFO]  * Camera name: zed
[zed_wrapper-2] 1650547554.882833078 [zed.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1650547554.882901849 [zed.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1650547554.883120037 [zed.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1650547554.883864781 [zed.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1650547554.884008213 [zed.zed_node] [INFO]  * Camera framerate: 15
[zed_wrapper-2] 1650547554.884271811 [zed.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1650547554.884352071 [zed.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1650547554.884630838 [zed.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1650547554.884885892 [zed.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1650547554.884997482 [zed.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-2] 1650547554.885119921 [zed.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1650547554.885196821 [zed.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-2] 1650547554.885355933 [zed.zed_node] [INFO]  * [DYN] Image downsample factor: 1 
[zed_wrapper-2] 1650547554.885531655 [zed.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-2] 1650547554.885687215 [zed.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-2] 1650547554.885850200 [zed.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-2] 1650547554.885986687 [zed.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-2] 1650547554.886111878 [zed.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-2] 1650547554.886354451 [zed.zed_node] [INFO]  * [DYN] Gamma: 8
[zed_wrapper-2] 1650547554.886485690 [zed.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[zed_wrapper-2] 1650547554.886628258 [zed.zed_node] [INFO]  * [DYN] Exposure: 80
[zed_wrapper-2] 1650547554.886733448 [zed.zed_node] [INFO]  * [DYN] Gain: 80
[zed_wrapper-2] 1650547554.886838221 [zed.zed_node] [INFO]  * [DYN] Auto White Balance: TRUE
[zed_wrapper-2] 1650547554.887053945 [zed.zed_node] [INFO]  * [DYN] White Balance Temperature: 42
[zed_wrapper-2] 1650547554.887183072 [zed.zed_node] [INFO]  * Video QoS History: KEEP_LAST
[zed_wrapper-2] 1650547554.887314215 [zed.zed_node] [INFO]  * Video QoS History depth: 1
[zed_wrapper-2] 1650547554.887461263 [zed.zed_node] [INFO]  * Video QoS Reliability: RELIABLE
[zed_wrapper-2] 1650547554.887592758 [zed.zed_node] [INFO]  * Video QoS Durability: VOLATILE
[zed_wrapper-2] 1650547554.887629176 [zed.zed_node] [INFO] *** DEPTH parameters ***
[... cut ...]
[zed_wrapper-2] 1650547556.719481492 [zed.zed_node] [INFO]  * Translation: {0.001,0.000,-0.015}
[zed_wrapper-2] 1650547556.719538359 [zed.zed_node] [INFO]  * Rotation: {0.000,-2.865,0.000}
[zed_wrapper-2] 1650547556.819683623 [zed.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1650547556.819855024 [zed.zed_node] [INFO]  * T: [-1.66741e-19,0.06,0.015]
[zed_wrapper-2] 1650547556.819978743 [zed.zed_node] [INFO]  * Q: [0,0.0249967,0,0.999687]
[zed_wrapper-2] 1650547556.961484857 [zed.zed_node] [INFO] Odometry aligned to last tracking pose

Myzhar, 2022-04-21:
Have you added this line?
`            publish_tf:                 false`
that's not in the right place. It must be in the `pos_tracking:` sub-group of parameters. You put it in the `general` subgroup.

You must be aware of what you are doing, otherwise what you do does not work as expected.

AndreV84, 2022-04-21:
@Myzhar Thank you for pointing out!
However after I added the false argument in the correct place
we got the different errors
```
ros2 run nvblox_ros nvblox_node --ros-args -r depth/camera_info:=/zed/zed_node/depth/camera_info -r depth/image:=/zed/zed_node/depth/depth_registered -r  color/image:=/zed/zed_node/rgb/image_rect_color -r color/camera_info:=/zed/zed_node/rgb/camera_info -p global_frame:=map
I0421 09:42:46.078943 19276 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 09:42:46.079536 19276 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 09:42:46.079569 19276 layer_cake_impl.h:31] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE , and memory_type: kDevice to LayerCake.
I0421 09:42:46.079596 19276 layer_cake_impl.h:31] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE , and memory_type: kDevice to LayerCake.
[INFO] [1650548566.090830837] [nvblox_node]: Outputting results (as requested) to: 
[INFO] [1650548566.091851192] [nvblox_node]: Started up nvblox node in frame map and voxel size 0.05
[INFO] [1650548566.203570554] [nvblox_node]: RGB frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650548566.261976289] [nvblox_node]: Depth frame statistics: 
avg=nan, min=nan, max=nan, std_dev=nan, count=0
[INFO] [1650548566.262242034] [nvblox_node]: Timing statistics: 
NVBlox Timing
-----------
ros/color	      1	00.000003	(00.000003 +- 00.000000)	[00.000003,00.000003]
ros/total	      2	00.000017	(00.000008 +- 00.000007)	[00.000001,00.000016]

Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
```
while from zed node the outputs are as follows
```
ros2 launch zed_wrapper zed.launch.py 
[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2022-04-21-09-42-38-132546-agx-19216
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: .
[INFO] [robot_state_publisher-1]: process started with pid [19219]
[INFO] [zed_wrapper-2]: process started with pid [19221]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed_base_link had 1 children
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[... cut ...]
[zed_wrapper-2] 1650548560.852511146 [zed.zed_node] [INFO]  * Translation: {0.001,0.000,-0.015}
[zed_wrapper-2] 1650548560.852568622 [zed.zed_node] [INFO]  * Rotation: {0.000,-2.865,0.000}
[zed_wrapper-2] 1650548560.952829677 [zed.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1650548560.952969942 [zed.zed_node] [INFO]  * T: [-1.66741e-19,0.06,0.015]
[zed_wrapper-2] 1650548560.953047419 [zed.zed_node] [INFO]  * Q: [0,0.0249967,0,0.999687]
[zed_wrapper-2] 1650548561.109792136 [zed.zed_node] [INFO] Odometry aligned to last tracking pose

AndreV84, 2022-04-21:
now zed node output shows rather 
```
[zed_wrapper-2] 1650548558.938777224 [zed.zed_node] [INFO]  * Broadcast Odometry TF: FALSE
```
than the previous output before adding the false argument which was
```
 [INFO] * Broadcast Pose TF: TRUE

AndreV84, 2022-04-21:
```
ros2 run tf2_tools view_frames.py
[INFO] [1650549111.133168235] [view_frames]: Listening to tf data during 5 seconds...
[INFO] [1650549116.160482297] [view_frames]: Generating graph in frames.pdf file...
[INFO] [1650549116.167002047] [view_frames]: Result:tf2_msgs.srv.FrameGraph_Response(frame_yaml="zed_base_link: \n  parent: 'base_link'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_camera_center: \n  parent: 'zed_base_link'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_left_camera_frame: \n  parent: 'zed_camera_center'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_left_camera_optical_frame: \n  parent: 'zed_left_camera_frame'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_right_camera_frame: \n  parent: 'zed_camera_center'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\nzed_right_camera_optical_frame: \n  parent: 'zed_right_camera_frame'\n  broadcaster: 'default_authority'\n  rate: 10000.000\n  most_recent_transform: 0.000000\n  oldest_transform: 0.000000\n  buffer_length: 0.000\n")

```
[frames.pdf](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/files/8531550/frames.pdf)

AndreV84, 2022-04-21:
now the false argument impacted the output of the command below
```
 ros2 run tf2_ros tf2_echo map zed_left_camera_frame
[INFO] [1650549405.588226037] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650549407.501974300] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650549408.501976058] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650549409.502040952] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650549410.502199513] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[INFO] [1650549412.501943503] [tf2_echo]: Waiting for transform map ->  zed_left_camera_frame: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
^C[INFO] [1650549413.334958408] [rclcpp]: signal_hand

AndreV84, 2022-04-21:
so as a user I am getting either this from the app
```
[ERROR] [1650550823.508767149] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.535148770] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.535341069] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.734482134] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.734647327] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.734713443] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.934379946] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.934567252] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.934637592] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550823.934697084] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550824.201481679] [nvblox_node]: Failed to transform color image.
[ERROR] [1650550824.201712956] [nvblox_node]: Failed to transform color image.
```
or If I add the proposed :false modification,
I will get this
```
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
         at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
```
if we disable the broadcast of TF as @Myzhar  proposed

AndreV84, 2022-04-21:
@helenol you may download recorded rosbag using the url below
https://andrev.file.core.windows.net/nvidiazed/rosbag2_2022_04_21-11_52_39_0.db3?sp=rl&st=2022-04-21T16:29:52Z&se=2022-04-22T16:29:52Z&sv=2020-08-04&sig=71lQBfs1WLlkAUYFPQ5yYyAJDen8jjmaoPozTsEFZU4%3D&sr=f

AndreV84, 2022-04-21:
@helenol _WERE YOU ABLE TO DOWNLOAD THE ROSBAG_?

cjh1995-ros, 2022-05-22:
Did you solve the problem? I am also trying to use realsense d435i and ZED2, but I have encountered the same problem as you.

 For the d435i, it looks like this: 
[image]

 In the case of ZED2, the depth mode was changed from performance to nueral, so the picture changed as follows.  
[image]
[image]

 But I still don't think I can get the color image from ZED2. If you solved the problem, please share it!

cjh1995-ros, 2022-05-22:
In the picture above, the last image is a picture of both zed's pcl and nvblox mesh. The mesh is not colored.

AndreV84, 2022-05-22:
@cjh1995-ros 
Thank you for following up!
Unfortunately I was not able to get the color from ZED.
@helenol  @Myzhar  used to try helping out with that but they seem gave up on supporting it

cjh1995-ros, 2022-05-22:
@AndreV84 
Thank you for your reply.
Too bad. I also wanted to use this library easily, but I still have a long way to go. 
From now on, I'm going to touch whether the parameters are important or not. If you have any experience, can you share it with me?

AndreV84, 2022-05-22:
@cjh1995-ros 
Nihao
Xie Xie

- ros nvblox parameters?

- ros zed  node parameters?
-  both? 
- neither of the two?

cjh1995-ros, 2022-05-22:
First, I'm not a Chinese. I'm from Korean. LOL

Both of them! 
In nvblox param folder, I saw the color hz parameter. So next Monday, I'm going to test that. 

Thank you for reply!

AndreV84, 2022-05-22:
haha,
we shall see
I do not even know where is nvblox param folder at this github repository
However we could use ros topic hz to show color topic information in terminal e.g. for diagnostics as in
```
ros2 topic hz /zed/zed_node/rgb/image_rect_color

[5 more replies]

closed 2022-08-01 by hemalshahNV