NVIDIA-ISAAC-ROS/isaac_ros_nvblox #9: Native installation won't work

FPSychotic, 2022-04-07:
I'm working with a native installation in x64 20.04, d435i. As AI modules, nvblox and nav2 won't work, as it sends errors similar to this one, doesn't matter what topics a put, if use / symbol or not as /camera/...
[nvblox_node-2]     @     0x55d6f42827be _start
[ERROR] [nvblox_node-2]: process has died [pid 43249, exit code -11, cmd '/home/igcs/isaac_ws/install/nvblox_ros/lib/nvblox_ros/nvblox_node --ros-args --params-file /home/igcs/isaac_ws/install/nvblox_nav2/share/nvblox_nav2/params/igcs_nvblox.yaml --params-file /tmp/launch_params_s3ofgvz8 -r depth/image:=camera/depth/image_rect_raw -r depth/camera_info:=camera/depth/camera_info -r color/image:=camera/color/image_raw -r color/camera_info:=camera/color/camera_info'].
[rviz2-1] [INFO] [1649357703.312523677] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1649357703.312730329] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1649357703.350449625] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1649357703.518772294] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1649357703.533156402] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1649357703.551877001] [rviz2]: Stereo is NOT SUPPORTED

it is like hardcoded to it only works only in Isaac Sim. I guess in the nvblox_node file. Little to little I was removing error after changing things, but this ends here. It looks Nav2 and vBlox only works in Sim I guess.

helenol, 2022-04-11:
Could be the same error as this one: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/issues/10

It's a bug with uint16 (most depth cameras) images vs float64 (isaac sim depth images), which is why you observe it working only with Isaac Sim. There's a quick one-line fix that works that we'll try to push out today.

helenol, 2022-05-10:
Fixed with #17 presumably. :) Closing. Please re-open if it's not fixed.

closed 2022-05-10 by helenol