NVIDIA-ISAAC-ROS/isaac_ros_nvblox #10: not able to visualize or see any data

indsy123, 2022-04-08:
Hi, I was trying to use nvblox ros with ubuntu 18.04 and realsene d455 camera. I installed ros-foxy from source, installed isaac_nvblox_ros and its dependencies along with realsense2_ros.  I think I supplied correct camera topics/transform and can run both but have two problems. 
1. As is, the nvblox_ros stops with SIGSEGV. I noticed that thrust::transform function in conversions.cu that divides the depth values by 1000 in RosConverter::depthImageFromImageMessage() is the cause (because I can use print commands up to that point) . The full error I get is: ***aborted at 1649440186 (unix time) try date -d 1649440186 if you are using GNU date.. 0x0 unkonwn, SIGSEGV received by PID 11109(TID 0x7fb5aba010) from PID 33530880; stack trace ***

2. I suppressed that line (thrust::transform()) just to see what happens. Now it runs and If I do "ros2 topic hz /nvblox_node/mesh" it seems to publish on to that topic. Same is with the pointcloud topic. However, there is no data in the fields when I echo the topics. I cant see it in rviz because I get the error:  " [rviz2]: could not load pixmap package://nvblox_ros_plugin/icons/classes/NvbloxMesh.png". I do have sourced the ros and nvblox bash files before running rviz. 

Can anyone please help with this or let me know if you need more information. Note that I am not running nav2_bringup and have suppressed the corresponding node in the carter_sim launch file. have also changed global frame in the nvblox-yaml file as per what I have from realsense2_ros. Just one thing, the ubuntu 18.04 that I have has jetpack 4.6 in case that is the root cause.

helenol, 2022-04-11:
@indsy123 
Hey thanks a ton for the report! We'll push a fix to this shortly (hopefully today) but there's a super quick hack fix you can do in the code: in `nvblox_ros/src/lib./cuda/conversions.cu` at around line 255, set:
```
bool kUseCuda = false;
```

This should fix the seg-fault. Sorry, this code-path ended up untested somehow. With this fix it works for the RS on my Jetson. :)

helenol, 2022-04-11:
For (2) you can ignore the rviz warning (just doesn't show up the icon in rviz) but you can do `git lfs pull` within the repo to get the .png. :)

indsy123, 2022-04-11:
Hi @helenol , 
Thanks for your reply. I could get it working with the quick fix you suggested and also was able to see the topics in rviz despite that error about pixmap. I appreciate your help and will switch to the fix once its pushed.

AndreV84, 2022-04-14:
@helenol I can not see any outputs either
while trying to process depth camera outputs
what can be done about it?
Will that patch work with native install?

closed 2022-05-10 by helenol

AndreV84, 2022-05-10:
@helenol 
so what is the solution to get zed color presented?