NVIDIA-ISAAC-ROS/isaac_ros_nvblox #11: Unable to include header files, <tf2_eigen/tf2_eigen.h>

GokulEpiphany, 2022-04-13:
I'm unable to build nvblox from source on Jetson Xavier NX.
Built ROS2 foxy from source, Built Navigation2 packages, Realsense packages and Isaac Elbrus Slam ROS packages. Everything works with a problem.
Sourced ROS2, While trying to build nvblox, nvblox_ros is unable to locate tf2_eigen.h header file. Although, the dependency in CMakelists.txt is met and can also verify that the packages are found under ~/ros2_foxy/. Any thoughts on the issue? 

--- stderr: nvblox_ros                                                                                                                                                                         
CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_ros_lib".
This warning is for project developers.  Use -Wno-dev to suppress it.

In file included from /home/eloopai/nav2_ws/src/isaac_ros_nvblox/nvblox_ros/src/transformer.cpp:11:0:
/home/eloopai/nav2_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/transformer.hpp:15:10: fatal error: tf2_eigen/tf2_eigen.h: No such file or directory
 #include <tf2_eigen/tf2_eigen.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nvblox_ros_lib.dir/src/transformer.cpp.o] Error 1
make[1]: *** [CMakeFiles/nvblox_ros_lib.dir/all] Error 2
make: *** [all] Error 2
---

GokulEpiphany, 2022-04-13:
On digging deeper, this might be caused because of the way ros2 foxy was built from source (colcon build --symlink-install) and omission of dependencies on CMakeLists.txt of nvblox_ros on Line 80. Explicitly adding tf2_eigen and tf2_ros to libraries dependencies resolves this issue. This might not be problem if ros2 foxy was built with colcon build. I would recommend explicitly adding tf2_eigen and tf2_ros dependencies to the transformer.cpp library.

closed 2022-04-13 by GokulEpiphany

helenol, 2022-04-19:
Hey @GokulEpiphany thanks a lot for the update! We have a newer version that handles the dependencies a bit nicer in the works, hopefully we can push it public soon. :) There's also a bit of conflict with tf_eigen since we bring in our own version of eigen (that's newer and actually supports cuda "properly"). Glad you got this working in the meantime and we'll incorporate the feedback in the next update!