NVIDIA-ISAAC-ROS/isaac_ros_nvblox #13: Loading a previously reconstructed representation?

GokulEpiphany, 2022-04-23:
I see that there is a service to save the reconstructed mesh. Is there a way to load the saved mesh into rviz, given the initial pose? Out of the box?

helenol, 2022-04-26:
Hi @GokulEpiphany , excellent question. This unfortunately doesn't exist at the moment. :( Serialization of the full map (including mesh) is near the top of the "future features" list so you can expect it in an upcoming release. You'd be able to save the map and load all layers including the mesh in a future execution.

There's some short-term workarounds though: you can export the mesh as a .ply using the service, and write a small script using pcl to load & visualize the mesh: https://answers.ros.org/question/203244/visualize-a-ply-file-with-a-pointcloud-in-rviz/ (that's for pointcloud, but it *is* also possible for mesh, pcl will load the mesh and look at the Marker Array mesh type, but this won't be too trivial). 
rviz can also natively display multiple mesh file types (.obj according to the link below, and definitely .dae and .stl, unfortunately AFAIK not .ply), and you can publish a mesh marker just pointing to the file: https://answers.ros.org/question/9299/how-to-display-a-mesh-with-texture-in-rviz/ (this is for ROS1 but should work the same in ROS2). This might be the easiest way to start. You can convert the mesh to other filetypes using MeshLab, for example.

hemalshahNV, 2022-09-19:
@alexmillane Any comment on if the latest version of nvblox satisfies this feature request?

alexmillane, 2022-10-25:
This is now possible. The `nvblox_node` now exposes the `load_map` service which will load a map from disk (the whole map, not just the mesh). It will then publish the mesh.

I hope that this fits what you were looking for.

closed 2022-11-15 by swapnesh-wani-nvidia