NVIDIA-ISAAC-ROS/isaac_ros_nvblox #16: Why this error happened?

cjh1995-ros, 2022-05-09:
Hello, I am a student who is using this library because I want to do research on navigation using isaac sim. 

I found the following error during execution, and I wonder what kind of phenomenon this error was caused by. 

This is the youtube link of the following error.

https://youtu.be/Rfgtxt02Tc4

And error is below

- in isaac sim shell

2022-05-09 03:46:24 [394,582ms] [Warning] [omni.usd] Warning: in Orthonormalize at line 495 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/base/gf/matrix4d.cpp -- OrthogonalizeBasis did not converge, matrix may not be orthonormal.

2022-05-09 03:46:24 [394,634ms] [Warning] [omni.usd] Warning: in Orthonormalize at line 495 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/base/gf/matrix4d.cpp -- OrthogonalizeBasis did not converge, matrix may not be orthonormal.

2022-05-09 03:46:24 [394,634ms] [Warning] [omni.usd] Warning: in Orthonormalize at line 495 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/base/gf/matrix4d.cpp -- OrthogonalizeBasis did not converge, matrix may not be orthonormal.

^CSimulation App Shutting Down

- in rviz2
 value in the transform (-nan -nan -nan) (nan nan nan nan)
[planner_server-4]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[planner_server-4] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[planner_server-4]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[controller_server-3]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[controller_server-3]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[controller_server-3]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[controller_server-3]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[controller_server-3]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[controller_server-3] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[controller_server-3]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[recoveries_server-5]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[recoveries_server-5]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[recoveries_server-5]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[recoveries_server-5]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[recoveries_server-5]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[recoveries_server-5] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[recoveries_server-5]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[bt_navigator-6]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[bt_navigator-6]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[bt_navigator-6]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[bt_navigator-6]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[bt_navigator-6]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_camera_stereo_right" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[bt_navigator-6]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[component_container-10] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[component_container-10]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[component_container-10] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "carter_lidar" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[component_container-10]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[component_container-10] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "carter_camera_stereo_left" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
[component_container-10]          at line 278 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[... cut ...]
[bt_navigator-6] Error:   TF DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "right_wheel_link" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan -nan nan -nan)
[bt_navigator-6]          at line 293 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[bt_navigator-6] Error:   TF NAN_INPUT: Ignoring transform for child_frame_id "rear_pivot_link" from authority "Authority undetectable" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)

helenol, 2022-05-09:
@cjh1995-ros This looks like an Isaac Sim issue. The robot explodes through the ground ( :( ) and then the orientation quaternions are not normalized and everything crashes. I don't really know how commanding base velocities through nav2 can cause this issue, but we've also seen it a few times on our side :/ Hopefully it's fixed in future Isaac Sim releases.

Sorry we can't be more help!

ewirbel, 2022-05-19:
Hello @cjh1995-ros ! I am not sure this will work but you could try reducing the maximum joint velocities of the wheels and pivot primitives (/World/Carter_ROS/chassis_link/left_wheel and same with right_wheel and rear_pivot). This can be found in the property panels of those prims, under Advanced. Try something like 360 deg /s (see attached screenshot)
[image]

cjh1995-ros, 2022-05-19:
> Hello @cjh1995-ros ! I am not sure this will work but you could try reducing the maximum joint velocities of the wheels and pivot primitives (/World/Carter_ROS/chassis_link/left_wheel and same with [...]

Thank you for opinion. But now, that crash is not happened. The real problem is i dont know why...

ewirbel, 2022-05-19:
I am pretty sure this is a divergence in the physics engine that can occur on some collisions between the articulation links and the ground. Capping the velocity is a workaround that ensures that the divergence is stopped before it actually sends the robot to outer space :) It should be fixed in the next versions, but in the meantime this workaround should make sure that when it happens it does not break your scene, and should not affect regular use cases

cjh1995-ros, 2022-05-19:
> I am pretty sure this is a divergence in the physics engine that can occur on some collisions between the articulation links and the ground. Capping the velocity is a workaround that ensures that [...]

I agree with that. May be, i push Isaac sim too hard. So it seems to be okay to close issue!