NVIDIA-ISAAC-ROS/isaac_ros_nvblox #19: How to reuse saved map?

cjh1995-ros, 2022-06-08:
A typical navgiation is to load a map using an image file and a yaml file that fills in its information. 

However, your package saves point cloud data, so I wrote a question wondering how to use this data for navigation again.

helenol, 2022-06-08:
Hey, serialization/reloading this is not currently supported but is coming in a future release :) The current state is meant to be used for local navigation - so you'd build up a new map next time you start up the robot.

cjh1995-ros, 2022-06-09:
Thanks for letting me know! Hope that comes fast!

muratkoc503, 2022-09-15:
Hi, when saving feature come nearly? 2 or 4 month? or long time?

cjh1995-ros, 2022-09-15:
@muratkoc503 I read the paper and it seems that it was not intended to save the map using this package. NVIDIA also recommends using it with Elbrus SLAM, so I'm thinking of the way using the two together. I'd appreciate it if you could tell me if you have any other opinions.

closed 2022-09-15 by cjh1995-ros