NVIDIA-ISAAC-ROS/isaac_ros_nvblox #24: Clearing voxels beyond a radius

Veng97, 2022-07-06:
I don't know if this feature already exists within the core library, but is there a method to prune 'far away' voxels or map blocks? :)

helenol, 2022-07-14:
Hey! Sorry for the slow response, this functionality isn't in yet but we'll add it to the list for the next release! All the blocks are independent so it's as easy as deleting the blocks that are out of range, which is functionality we'll need to add to the core library.

hemalshahNV, 2022-09-19:
@alexmillane Any comment on if the latest version of nvblox supports this the feature they way it was described here?

alexmillane, 2022-10-25:
This is now in the library and configurable through the ROS param `map_clearing_radius_m`

Veng97, 2022-10-25:
Awesome!

closed 2022-11-15 by swapnesh-wani-nvidia