NVIDIA-ISAAC-ROS/isaac_ros_nvblox #29: Rgbd mapper and integrator parameters for the Intel realsense cameras

arainbilal, 2022-08-16:
Hi @helenol,
From your experience using the Intel realsense cameras in indoor environments, just wondering if you may be able to help elaborate on the following rgbd mapper parameters, which I am trying to tune to get the best possible results using the D455/D435 cameras to compare with the Sun3D dataset.  

Parameter | value | Remarks 
- voxel size |  0.05 | The size of the tsdf voxels
- max_integration_distance | 4.0 | Max distance in the depth frame in meter; parameter is defined in projective_integrator_base.h
- truncation_distance_vox | 4.0 | Truncation distance for the TSDF; (In the paper it is set to 4*v)
- integrator max weight | 1e4 | The upper limit for the weight assigned to a voxel
- integrator min weight | 1e-4 | the lower limit for the weight assigned to  a voxel
- weld vertices are set to false.
 
In addition to the integrator parameter, I have used the following:
- esdf is set to false and only updating mesh using converter_.meshMessageFromMeshBlocks(). therefore slice height and min/max height is not used.

Where possible, I have looked into the source code and the readme files but another look or more explanation would be greatly appreciated. The example provided using the Sun3D dataset using Asus Xiton Pro seems to produce better results than the realsense and I am not sure why this is the case. Note that the depth image encoding is 16UC1 from the realsense sensor and therefore the converter performs the division. Thanks in advance.

alexmillane, 2022-08-25:
The quality of the map depends on the quality of the input depth maps. Do you have the projector enabled on your realsense.

We will release example launchfiles and code for building reconstructions from a realsense camera in the next couple of weeks. We're in the final stages of preparing this.

It's not ideal to have to wait a couple of weeks, however, but once we release this I think it will be easier to talk about an example application we're both running.

alexmillane, 2022-09-02:
Our realsense example is now live. Let us know if you find it useful

arainbilal, 2022-09-15:
Thanks, @alexmillane. Question about the realsense splitter node. Emitter is enabled for the depth image (https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/2c6686676a758b16677b7a143a018a4e29630db3/nvblox_examples/realsense_splitter/src/realsense_splitter_node.cpp#L136), but emitter is off for the localisation (https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/2c6686676a758b16677b7a143a018a4e29630db3/nvblox_examples/realsense_splitter/src/realsense_splitter_node.cpp#L120). If emitter is enabled, infra1/2 from the splitter will not publish? Which means, the pose is not available (https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/2c6686676a758b16677b7a143a018a4e29630db3/nvblox_examples/nvblox_examples_bringup/launch/nvblox_vslam_realsense.launch.py#L91). 

Thanks in advance for your help.

swapnesh-wani-nvidia, 2022-10-10:
@alexmillane or @helenol do you have any updates on this?

alexmillane, 2022-10-11:
Our example configures the realsense to turn the emitter on and off on alternating frames. Half of the frames for to VSLAM (emitter off) and half go to nvblox (emitter on). Using this setup nvblox and VSLAM are able to run simultaneously :).

arainbilal, 2022-10-26:
Please close this issue. Thanks.

closed 2022-10-26 by arainbilal