NVIDIA-ISAAC-ROS/isaac_ros_nvblox #34: Odom frame does not exist when running Tutorial with realsense, VSLAM and nvblox

chivas1000, 2022-10-04:
Hi
I'm implementing nvblox and VSLAM on my robots to do a navigation application

The installation is good and I can run the examples of nvblox bag example and VSLAM realsense example

but Got Error When running the nvblox realsense and NVSLAM examples
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-nvblox-vslam-realsense.md

here are error logs:
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[rviz2-5] [INFO] [1664879434.204059913] [rclcpp]: signal_handler(signum=2)
[rviz2-5] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-5]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-5]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[component_container-1] [INFO] [1664879436.427147138] [rclcpp]: signal_handler(signum=2)
[component_container-1] [INFO] [1664879436.666174595] [rclcpp]: signal_handler(signum=2)
[component_container-1] [INFO] [1664879436.827491359] [rclcpp]: signal_handler(signum=2)
[component_container-1] [INFO] [1664879437.023495947] [rclcpp]: signal_handler(signum=2)
[component_container-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[component_container-1]   what():  parameter 'accel_fps' has invalid type: cannot undeclare an statically typed parameter

I can confirm the camera are running since the topic can output images.
Does anyone have ideas on how this could be solve?
BTW I'm new to docker, why does the rqt_graph doesn't work in the container although installed it?

Much appreciated if anyone could help.

chivas1000, 2022-10-05:
by the way, I'm using jetson AGX Xavier with jetpack 5.0.2 with realsense D435i

chivas1000, 2022-10-05:
looking at the graph it seems that the VSLAM node doesn't open
[image]

Error Log more specific after launching the script

admin@ubuntu:~$ ros2 launch nvblox_examples_bringup nvblox_vslam_realsense.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2022-10-05-07-44-54-315962-ubuntu-11348
[INFO] [launch]: Default logging verbosity is set to INFO
/workspaces/isaac_ros-dev/install/nvblox_examples_bringup/share/nvblox_examples_bringup/config/nvblox_vslam_realsense.rviz
[INFO] [component_container-1]: process started with pid [11359]
[INFO] [isaac_ros_visual_slam-2]: process started with pid [11361]
[INFO] [nvblox_node-3]: process started with pid [11363]
[INFO] [static_transform_publisher-4]: process started with pid [11365]
[INFO] [rviz2-5]: process started with pid [11367]
[static_transform_publisher-4] [WARN] [1664955896.323346218] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-4] [INFO] [1664955896.513530017] [static_transform_publisher_TTBBBeHEbhxSJuCp]: Spinning until stopped - publishing transform
[static_transform_publisher-4] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-4] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-4] from 'base_link' to 'camera_link'
[component_container-1] [INFO] [1664955896.622333117] [realsense_container]: Load Library: /workspaces/isaac_ros-dev/install/realsense2_camera/lib/librealsense2_camera.so
[component_container-1] [INFO] [1664955896.796502685] [realsense_container]: Found class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[component_container-1] [INFO] [1664955896.796745681] [realsense_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[rviz2-5] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-admin'
[nvblox_node-3] I1005 07:44:56.875982 11363 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9TsdfVoxelEEE, voxel_size: 0.05, and memory_type: kDevice to LayerCake.
[nvblox_node-3] I1005 07:44:56.880472 11363 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_10ColorVoxelEEE, voxel_size: 0.05, and memory_type: kDevice to LayerCake.
[nvblox_node-3] I1005 07:44:56.880540 11363 layer_cake_impl.h:32] Adding Layer with type: N6nvblox15VoxelBlockLayerINS_9EsdfVoxelEEE, voxel_size: 0.05, and memory_type: kDevice to LayerCake.
[nvblox_node-3] I1005 07:44:56.880581 11363 layer_cake_impl.h:32] Adding Layer with type: N6nvblox10BlockLayerINS_9MeshBlockEEE, voxel_size: 0.05, and memory_type: kDevice to LayerCake.
[nvblox_node-3] [INFO] [1664955896.969680911] [nvblox_node]: Outputting results (as requested) to: /home/nvidia/data/meshes/
[nvblox_node-3] [INFO] [1664955896.970138611] [nvblox_node]: Started up nvblox node in frame odom and voxel size 0.05
[isaac_ros_visual_slam-2] [INFO] [1664955897.007133002] [visual_slam_node]: Elbrus version: 10.0
[component_container-1] [INFO] [1664955897.187497561] [camera.camera]: RealSense ROS v3.2.3
[component_container-1] [INFO] [1664955897.187686248] [camera.camera]: Built with LibRealSense v2.51.1
[component_container-1] [INFO] [1664955897.187767150] [camera.camera]: Running with LibRealSense v2.51.1
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/realsense_container'
[component_container-1] [INFO] [1664955897.220192951] [realsense_container]: Load Library: /workspaces/isaac_ros-dev/install/realsense_splitter/lib/librealsense_splitter_component.so
[component_container-1] [INFO] [1664955897.251181709] [realsense_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container-1] [INFO] [1664955897.251414592] [realsense_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container-1] [INFO] [1664955897.333858306] [camera.realsense_splitter_node]: Creating a RealsenseSplitterNode().
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/realsense_splitter_node' in container '/realsense_container'
[component_container-1] [INFO] [1664955898.031412018] [camera.camera]: Device with serial number 135122070687 was found.
[component_container-1] 
[component_container-1] [INFO] [1664955898.031640388] [camera.camera]: Device with physical ID 2-4.1.3-8 was found.
[component_container-1] [INFO] [1664955898.031755437] [camera.camera]: Device with name Intel RealSense D435I was found.
[component_container-1] [INFO] [1664955898.033193375] [camera.camera]: Device with port number 2-4.1.3 was found.
[component_container-1] [INFO] [1664955898.033492919] [camera.camera]: Device USB type: 3.2
[component_container-1] [INFO] [1664955898.088029646] [camera.camera]: getParameters...
[component_container-1] [INFO] [1664955898.183163630] [camera.camera]: setupDevice...
[component_container-1] [INFO] [1664955898.183306489] [camera.camera]: JSON file is not provided
[component_container-1] [INFO] [1664955898.183403329] [camera.camera]: Device Name: Intel RealSense D435I
[component_container-1] [INFO] [1664955898.183498024] [camera.camera]: Device physical port: 2-4.1.3-8
[component_container-1] [INFO] [1664955898.183561325] [camera.camera]: Device FW version: 05.13.00.50
[component_container-1] [INFO] [1664955898.183641364] [camera.camera]: Device Product ID: 0x0B3A
[... cut ...]
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[component_container-1]  05/10 07:45:05,396 WARNING [281472225241392] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[nvblox_node-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[nvblox_node-3]          at line 93 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp

alexmillane, 2022-10-07:
Hi there. Thank you for the message and for your interest in nvblox.

I think your diagnosis is correct, the issue is that vslam hasn't started.

The (annoying) messages on the console are because, nvblox checks at a 100Hz to see if the sensor pose is available and can't find the transform from 'odom' to the camera.

Could you just check a few things?
- vslam is successfully built
- just double check it's _definately_ not running by running: ros2 node list

chivas1000, 2022-10-08:
> Hi there. Thank you for the message and for your interest in nvblox.
>
> I think your diagnosis is correct, the issue is that vslam hasn't started.
>
> The (annoying) messages on the console are because, [...]

Hi @alexmillane, thanks for your reply

I ran it over again with the **nvblox_vslam_realsense.launch.py** and seem that my camera is outputting /camera/color/image_raw, but not outputting images for /camera/realsense_splitter_node/output/infra_1, which is the one vlam subscribed.
[image]

and /camera/depth/image_rect_raw no output too

but seeing my realsense-viewer, it works properly.
[image]

and rqt_graph seems that vslam_node and nvblox_node are up and topics subscrition are right.
[image]

Here is my ros2 node list, it seems that these two nodes are running but maybe this time it has no image input so it does not work
agilex@ubuntu:~$ ros2 node list
/camera/camera
/camera/realsense_splitter_node
/launch_ros_29486
/nvblox_node
/realsense_container
/rviz
/static_transform_publisher_wErZdyhdFXtjHQqz
/transform_listener_impl_aaaaca406500
/transform_listener_impl_aaaaee2c8f30
/transform_listener_impl_aaab12b8eb50
/visual_slam_node

for your question that if the vslam is successfully built
I ran the example of isaac_ros_visual_slam_realsense.launch.py and it works.
[image]

I want to only running the realsense node on the nvblox_vslam_realsense.launch.py by only leaving the realsense related lines, but there are no output both for topic echo and rviz2
[image]

alexmillane, 2022-10-10:
Great that you got it working.

To me, it appears that something is wrong with the projector toggling. 

One potential issue with removing the splitter is that, if the projector is on, the projector pattern with cause VSLAM tracking performance to degrade.

Could you have a look at the IR images and see if you see the projector pattern? They appear as spots.

One thing you could also try is adding 

`'image_qos': 'SENSOR_DATA'`

to the VSLAM parameters in the launchfile. This should have been in there, and it is a bug that it's not. Thank you for finding that. We will correct this shortly.

Regarding the ESDF map: the 2D ESDF slice should already be published. Try to visualize it in RVIZ and see if you can.

We're using the 2D slice here at nvidia to navigate ground robots.

chivas1000, 2022-10-10:
> 

Hi @alexmillane,
Thanks for your reply and  I learned from you that if the IR emitting pattern can be shut down it will increase VSLAM performance, But sadly my splitter still not working, it only show messages for the depth images but not two infrared cams, so the VSLAM node can't get images from splitter.

In order to get the splitter work, I've try to just launch the realsense, splitter in a container and see the output:

admin@ubuntu:/workspaces/isaac_ros-dev$ ros2 launch nvblox_examples_bringup realsense_2.launch.py 
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2022-10-10-12-37-42-589962-ubuntu-17212
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [17235]
[INFO] [static_transform_publisher-2]: process started with pid [17237]
[static_transform_publisher-2] [WARN] [1665405463.696407382] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1665405463.825379821] [static_transform_publisher_ZmundZFhUAx2yf3z]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'base_link' to 'camera_link'
[component_container-1] [INFO] [1665405463.844361887] [realsense_container]: Load Library: /workspaces/isaac_ros-dev/install/realsense2_camera/lib/librealsense2_camera.so
[component_container-1] [INFO] [1665405463.978694639] [realsense_container]: Found class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[component_container-1] [INFO] [1665405463.978952568] [realsense_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<realsense2_camera::RealSenseNodeFactory>
[component_container-1] [INFO] [1665405464.175236779] [camera.camera]: RealSense ROS v3.2.3
[component_container-1] [INFO] [1665405464.175498452] [camera.camera]: Built with LibRealSense v2.51.1
[component_container-1] [INFO] [1665405464.175620761] [camera.camera]: Running with LibRealSense v2.51.1
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/realsense_container'
[component_container-1] [INFO] [1665405464.207149885] [realsense_container]: Load Library: /workspaces/isaac_ros-dev/install/realsense_splitter/lib/librealsense_splitter_component.so
[component_container-1] [INFO] [1665405464.217913843] [realsense_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container-1] [INFO] [1665405464.218159451] [realsense_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container-1] [INFO] [1665405464.247750205] [camera.realsense_splitter_node]: Creating a RealsenseSplitterNode().
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/realsense_splitter_node' in container '/realsense_container'
[component_container-1] [INFO] [1665405464.905777779] [camera.camera]: Device with serial number 135122070687 was found.
[component_container-1] 
[component_container-1] [INFO] [1665405464.910863908] [camera.camera]: Device with physical ID 2-4.1.3-5 was found.
[component_container-1] [INFO] [1665405464.910997576] [camera.camera]: Device with name Intel RealSense D435I was found.
[component_container-1] [INFO] [1665405464.912858537] [camera.camera]: Device with port number 2-4.1.3 was found.
[component_container-1] [INFO] [1665405464.913038159] [camera.camera]: Device USB type: 3.2
[component_container-1] [INFO] [1665405464.950806828] [camera.camera]: getParameters...
[component_container-1] [INFO] [1665405464.999962804] [camera.camera]: setupDevice...
[component_container-1] [INFO] [1665405465.001247776] [camera.camera]: JSON file is not provided
[component_container-1] [INFO] [1665405465.001374693] [camera.camera]: Device Name: Intel RealSense D435I
[component_container-1] [INFO] [1665405465.001445479] [camera.camera]: Device physical port: 2-4.1.3-5
[component_container-1] [INFO] [1665405465.001506889] [camera.camera]: Device FW version: 05.13.00.50
[component_container-1] [INFO] [1665405465.001565547] [camera.camera]: Device Product ID: 0x0B3A
[component_container-1] [INFO] [1665405465.001623629] [camera.camera]: Enable PointCloud: Off
[component_container-1] [INFO] [1665405465.001682351] [camera.camera]: Align Depth: Off
[component_container-1] [INFO] [1665405465.001738257] [camera.camera]: Sync Mode: Off
[component_container-1] [INFO] [1665405465.001854997] [camera.camera]: Device Sensors: 
[component_container-1] [INFO] [1665405465.197056738] [camera.camera]: Stereo Module was found.
[component_container-1] [INFO] [1665405465.308215310] [camera.camera]: RGB Camera was found.
[component_container-1] [INFO] [1665405465.309233937] [camera.camera]: Motion Module was found.
[component_container-1] [INFO] [1665405465.309495707] [camera.camera]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[component_container-1] [INFO] [1665405465.309606526] [camera.camera]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[component_container-1] [INFO] [1665405465.309680609] [camera.camera]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[... cut ...]
[component_container-1]  10/10 12:42:34,037 WARNING [281473214404912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[component_container-1]  10/10 12:42:54,122 WARNING [281473214404912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

I see that there are some errors there, and I look into the splitter code,  is that seems like the splitter couldn't tell the realsense node to turn off but it said it cannot, so there are no output from splitter?

alexmillane, 2022-10-10:
Just to double check again, the infrared topics before the splitter are there? Correct?

So I'm having the same issue as you on one of my computers and I have made some progress in solving it. Could you post the results of the following three commands?

```
uname -r

dkms status

modinfo uvcvideo | grep "version:"
```

chivas1000, 2022-10-11:
> 

Yes, the  infrared topics before the splitter are there.
[image]

As to the command, I think you mean running inside the container, the dkms is first outputs "bash: dkms: command not found" and I apt installed it, and it has no output, so do the third commands.

inside the container:
admin@ubuntu:/workspaces/isaac_ros-dev$ uname -r
5.10.104-tegra
admin@ubuntu:/workspaces/isaac_ros-dev$ dkms status
admin@ubuntu:/workspaces/isaac_ros-dev$ modinfo uvcvideo | grep "version:"
modinfo: ERROR: Module alias uvcvideo not found.
admin@ubuntu:/workspaces/isaac_ros-dev$ 

for your convenience, I ran it outside the container too, and dkms is also not installed and no output after apt-get install:
agilex@ubuntu:~$ dkms status
agilex@ubuntu:~$ uname -r
5.10.104-tegra
agilex@ubuntu:~$ dkms status
agilex@ubuntu:~$ modinfo uvcvideo | grep "version:"
version:        1.1.1
srcversion:     187DBDC4F94931600634224
agilex@ubuntu:~$ 

Hope these information helps, thank you.

helenol, 2022-10-11:
Hi @chivas1000 , I think the issue is lacking the kernel patch outside the container to run the RealSense.
So one "easy" thing to try is to install the realsense drivers outside the container:
https://github.com/JetsonHacksNano/installLibrealsense/blob/master/installLibrealsense.sh
This should also bring in the `librealsense2-dkms` package which should bring in the kernel components and patches.

Another thing to try is to build the realsense drivers with ` -DFORCE_RSUSB_BACKEND=ON ` per this issue: https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

For newer kernels we found we needed to use a different work-around, detailed here: https://github.com/IntelRealSense/librealsense/issues/10439
Though the pre-compiled binaries are only available for amd64 (though anyone struggling with the same issues on x86 might find this link useful): https://github.com/mengyui/librealsense2-dkms/releases

chivas1000, 2022-10-11:
> Hi @chivas1000 , I think the issue is lacking the kernel patch outside the container to run the RealSense. So one "easy" thing to try is to install the realsense drivers outside the container: https://github.com/JetsonHacksNano/installLibrealsense/blob/master/installLibrealsense.sh This should also bring in the `librealsense2-dkms` package which should bring in the kernel components and patches.
> 
> Another thing to try is to build the realsense drivers with `-DFORCE_RSUSB_BACKEND=ON` per this issue: [IntelRealSense/librealsense#6964 (comment)](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049)
> 
> For newer kernels we found we needed to use a different work-around, detailed here: [IntelRealSense/librealsense#10439](https://github.com/IntelRealSense/librealsense/issues/10439) Though the pre-compiled binaries are only available for amd64 (though anyone struggling with the same issues on x86 might find this link useful): https://github.com/mengyui/librealsense2-dkms/releases

Thank you @helenol , but it seems that there are no avaliable realsense binary releases of jetpack 5.0.2 i.e. kernel 5.10.104-tegra, I'm trying to build it from source to see if it can solve this issue

chivas1000, 2022-10-12:
> Hi @chivas1000 , I think the issue is lacking the kernel patch outside the container to run the RealSense. So one "easy" thing to try is to install the realsense drivers outside the container: [...]

Hi @helenol, I managed to build and install realsense outside the container with steps of:
https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh
and your advice on putting -DFORCE_RSUSB_BACKEND=ON
https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

but after installing it, and open the container and build the colcon build nvblox again, it doesn't solve the issue

I supposed that your idea of kernel patch needs to be install is right and USB_BACKEND doesn't work since maybe only the kernel controls the depth camera advanced features.

But the kernel patch you gave are for kernel 5.13/15 amd64 and jetpack 5.0.2 is kernel 5.1.104, so I don't know if it could supports, should I build the kernel patch or just wait for the release?

Thanks for your help

alexmillane, 2022-10-13:
Hi @chivas1000. Somehow I missed your message about being on the jetson XD. I was giving advice assuming x86. Could you send me an email at amillane@nvidia.com?

alexmillane, 2022-10-25:
Hmmm yea good point, the dmks-based solution suggested by my troubleshooting is for x86 only.

Obviously for us at nvidia switching from a jetson to a nuc is not a great solution...

I'm running on the jetson quite frequently. So I'm trying to understand what's different about your system. I'm going to record a video of things running that might help you... I'll get back to you later this week.

ripdk12, 2022-10-28:
Just chiming in, as I have almost the exact same setup, just with a Xavier NX instead, and am experiencing the same issues. @chivas1000 can you give a slightly more detailed description of how you fixed the issue? Also did you get the rqt_graph working inside the container? Thanks :D

chivas1000, 2022-10-28:
> Just chiming in, as I have almost the exact same setup, just with a Xavier NX instead, and am experiencing the same issues. @chivas1000 can you give a slightly more detailed description of how you fixed the issue? Also did you get the rqt_graph working inside the container? Thanks :D

Hi @ripdk12 ,
I've just switched to nuc to avoid the splitter not working issue,
But as for the xavier, I assume that you can just build librealsense from source outside the container for this 
https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta
at step 2
and then go through the docker setup of this repo to see if it can output.

ripdk12, 2022-10-29:
> Hi @ripdk12 , I've just switched to nuc to avoid the splitter not working issue, But as for the xavier, I assume that you can just build librealsense from source outside the container for this [...]

I did this already and this fixed the odom frame problem, but am still experience issues, and switching to nuc is not an option in my case. I suppose I'll open an issue myself with more details, but thanks for the snappy response!

closed 2022-11-22 by hemalshahNV