NVIDIA-ISAAC-ROS/isaac_ros_nvblox #37: Waiting for service map_server/get_state on nav2 application

chivas1000, 2022-10-26:
Hi, I’m modifying the **carter_sim_elbrus.launch.py** example in nvblox_nav2 to do navigation in my robot, 
The application is ok at DP1.1 release before
but for DP2.0, after the configurinig is done, the terminal keep popping “waiting for server state”

Error Output:
[lifecycle_manager-10] [INFO] [1666756876.922715146] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-1] [INFO] [1666756876.922716862] [lifecycle_manager_map]: Waiting for service map_server/get_state...
[lifecycle_manager-1] [INFO] [1666756878.923016306] [lifecycle_manager_map]: Waiting for service map_server/get_state...
[lifecycle_manager-10] [INFO] [1666756878.923042882] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-1] [INFO] [1666756880.923268323] [lifecycle_manager_map]: Waiting for service map_server/get_state...
[lifecycle_manager-10] [INFO] [1666756880.923292501] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-10] [INFO] [1666756882.923515151] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-1] [INFO] [1666756882.923517381] [lifecycle_manager_map]: Waiting for service map_server/get_state...
[lifecycle_manager-1] [INFO] [1666756884.923782959] [lifecycle_manager_map]: Waiting for service map_server/get_state...
[lifecycle_manager-10] [INFO] [1666756884.923803742] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[nvblox_node-11] [INFO] [1666756885.119232012] [nvblox_node]: RGB frame statistics: 
[nvblox_node-11] avg=66.713883, min=55.955423, max=84.688353, std_dev=5.970719, count=13211
[nvblox_node-11] [INFO] [1666756885.463369156] [nvblox_node]: Depth frame statistics: 
[nvblox_node-11] avg=36.677327, min=13.215166, max=126.695587, std_dev=10.915946, count=24039
[nvblox_node-11] [INFO] [1666756885.463661005] [nvblox_node]: Timing statistics: 
[nvblox_node-11] NVBlox Timing
[nvblox_node-11] -----------
[nvblox_node-11] color/integrate

But the nvblox and vslam seem working as expect
[image]

so maybe is that the navigation stack doesn’t working since there are no availiable map input even though the nvblox does produced the map, so I assume that the nvblox has output esdf map but map server didn’t receive it

My launch files:

**carter_sim_elbrus_new.launch.py
(for this the I didn't modified the map server related lines)**

# Copyright (c) 2022, NVIDIA CORPORATION.  All rights reserved.
#
# NVIDIA CORPORATION and its licensors retain all intellectual property
# and proprietary rights in and to this software, related documentation
# and any modifications thereto.  Any use, reproduction, disclosure or
# distribution of this software and related documentation without an express
# license agreement from NVIDIA CORPORATION is strictly prohibited.

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import ComposableNodeContainer
from launch_ros.actions import Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():

    #simulation time, this would not be use in the our application
    use_sim_time = LaunchConfiguration('use_sim_time', default='False')

    #costmap param, need to be modify
    param_dir = LaunchConfiguration(
        'params_file',
        default=os.path.join(
            get_package_share_directory(
                'nvblox_nav2'), 'params', 'carter_nav2_new.yaml'
        ),
    )

[... 365 lines cut ...]
    global_frame: map
    robot_base_frame: base_link
    transform_tolerance: 0.2
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 0.2
    min_rotational_vel: 0.05
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

has anyone ran into this problem? and any advises are appreciated.
Thanks

chivas1000, 2022-10-27:
Sorry, the problem should be in my modifided carter_nav2.yaml that I tweaked with tolerance time and radius and acc_lim, or maybe some syntax problem

helenol, 2022-11-09:
Since the default yaml is working, I'm not sure how we can help here. Nothing you've changed stands out at something that should cause this issue, but this seems like a nav2 issue rather than an nvblox one.
Sorry we couldn't help more!

closed 2022-11-09 by helenol