NVIDIA-ISAAC-ROS/isaac_ros_nvblox #38: Garbage output running nvblox on Orin with Realsense D455

div-garg, 2022-10-26:
Nvblox seems to be working fine for me when I try the Isaac sim example but is giving me very bad outputs when running the realsense-based reconstruction example for nvblox. I am using a Orin Dev Kit with the latest tegra version and a Realsense D455 camera.

I have also gone through the troubleshoot page, and all the required metadata and topics are getting published to Rviz. I am not sure what the issue might be, any help will be appreciated. Thanks!

[image]

alexmillane, 2022-10-30:
Hi there. Thanks for your interesting in nvblox.

Does VSLAM appear to track smoothly? Some other users are reporting that VLSAM is saying `tracking is lost` on the terminal. Is that the case for you?

div-garg, 2022-10-31:
Yes, I am seeing `tracking is lost` error a lot in the terminal currently when moving the realsense unless I move it very slowly (~1 cm / sec). Another error I am seeing sometimes is `discarding message because the queue is full`. 

Let me know if there are any mitigations to the problems.

jaiveersinghNV, 2022-11-08:
Could you see if the solution from [this issue](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/54#issuecomment-1297800417) fixes your problem?

div-garg, 2022-11-08:
Yes, this fix worked! Thanks for addressing the issue.

closed 2022-11-08 by div-garg