NVIDIA-ISAAC-ROS/isaac_ros_nvblox #46: can nvblox ros use lidar as input?

chivas1000, 2022-11-29:
Hi I'm using nvblox to produce a costmap for my robot to do a navigation
can it inputs Lidar pointcloud data and still works?
also, if I use other lidar based SLAM as a state estimator, would it works?
since the rgbd and double stereo IR camera and RGB camera altogether takes up too much bandwidth
If nvblox can inputs lidar data and so do vslam, then It might be able to just using lidar data.
Is that right or I am missing something?
thanks for any reply.

alexmillane, 2022-11-30:
nvblox_ros accepts 3D LiDAR input. We are also agnostic as to where the pose estimate comes from, so an (accurate) 3D pose from LiDAR SLAM will work fine.

I am somewhat surprised you have bandwidth problems though. Are you using the realsense? What computer are you running on?

chivas1000, 2022-12-01:
> nvblox_ros accepts 3D LiDAR input. We are also agnostic as to where the pose estimate comes from, so an (accurate) 3D pose from LiDAR SLAM will work fine.
> 
> I am somewhat surprised you have bandwidth problems though. Are you using the realsense? What computer are you running on?

Hi @alexmillane,
thanks for the reply, and for the bandwidth, I'm using realsense d435i and now I'm transmitting these datas from one computer to another, so smaller bandwidth usage would be appreciated

another question is that can this nvblox map reuse as to the navigation application? since for the navigation loading a map which is already built is more convenient.

Thanks for your reply.

alexmillane, 2022-12-05:
Ah I see. Your bandwidth limitations now make sense.

The `nvblox_node` exposes a load map service that can be used to load an existing map for navigation.

closed 2022-12-14 by chivas1000

tugbakara, 2024-02-15:
> Ah I see. Your bandwidth limitations now make sense.
> 
> The `nvblox_node` exposes a load map service that can be used to load an existing map for navigation.

How can I use this map for navigation2? I couldnt find any example related to this

alexmillane, 2024-02-15:
We have docs about using this package for navigation on Nova Carter [here](https://nvidia-isaac-ros.github.io/robots/nova_carter.html).

To run it on your own robot, you'll have to adapt the nova carter launchfiles to your robot.

tugbakara, 2024-02-15:
> [here](https://nvidia-isaac-ros.github.io/robots/nova_carter.html).
I glanced this page but there does not exist isaac_ros-dev package in github and this nova carter tutorial launch files run from packe isaac_ros-dev.  That's why I cannot reach launch files or anyother to implement my robot