NVIDIA-ISAAC-ROS/isaac_ros_nvblox #47: Build fails

nakai-omer, 2022-11-29:
Hi,

Trying to build this packages (after running rosdep install), yields the following errors:
```
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:739:3: error: ‘Lidar’ was not declared in this scope
  739 |   Lidar lidar(lidar_width_, lidar_height_, lidar_vertical_fov_rad_);
      |   ^~~~~
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:748:56: error: ‘lidar’ was not declared in this scope
  748 |   if (!converter_.checkLidarPointcloud(pointcloud_ptr, lidar)) {
      |                                                        ^~~~~
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:758:21: error: ‘lidar’ was not declared in this scope
  758 |     pointcloud_ptr, lidar,
      |                     ^~~~~
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:764:12: error: ‘class nvblox::RgbdMapper’ has no member named ‘integrateLidarDepth’; did you mean ‘integrateDepth’?
  764 |   mapper_->integrateLidarDepth(pointcloud_image_, T_S_C, lidar);
      |            ^~~~~~~~~~~~~~~~~~~
      |            integrateDepth
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp: In member function ‘void nvblox::NvbloxNode::clearMapOutsideOfRadius(const string&, const rclcpp::Time&)’:
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:903:54: error: ‘class nvblox::RgbdMapper’ has no member named ‘clearOutsideRadius’
  903 |       std::vector<Index3D> blocks_cleared = mapper_->clearOutsideRadius(
      |                                                      ^~~~~~~~~~~~~~~~~~
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp: In member function ‘void nvblox::NvbloxNode::saveMap(std::shared_ptr<nvblox_msgs::srv::FilePath_Request_<std::allocator<void> > >, std::shared_ptr<nvblox_msgs::srv::FilePath_Response_<std::allocator<void> > >)’:
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:974:32: error: ‘class nvblox::RgbdMapper’ has no member named ‘saveMap’
  974 |   response->success = mapper_->saveMap(filename);
      |                                ^~~~~~~
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp: In member function ‘void nvblox::NvbloxNode::loadMap(std::shared_ptr<nvblox_msgs::srv::FilePath_Request_<std::allocator<void> > >, std::shared_ptr<nvblox_msgs::srv::FilePath_Response_<std::allocator<void> > >)’:
/home/amram/development/nakai/packages/nakai-onboard/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:994:32: error: ‘class nvblox::RgbdMapper’ has no member named ‘loadMap’
  994 |   response->success = mapper_->loadMap(filename);
      |                                ^~~~~~~
```

alexmillane, 2022-11-30:
To me, this looks like you haven't pulled the nvblox submodule.

Can you run:
```
git submodule update
```

nakai-omer, 2022-12-01:
@alexmillane Thanks a lot, I had it pulled, but it wasn't updated to the latest. Problem solved!

closed 2022-12-01 by nakai-omer