NVIDIA-ISAAC-ROS/isaac_ros_nvblox #48: Magic number in carter_sim.launch.py

james-yoo, 2022-12-23:
Hi,

Can someone explain to me the hardcoded value in the carter_sim.launch.py? Why this offset is needed?

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/4481418f69a7a785b7eeb36be96ee561989c7a3c/nvblox_nav2/launch/carter_sim.launch.py#L103

chivas1000, 2022-12-27:
this is a static tf transform
and it tells the coordinate transform between odom to map
you can refer to 
http://wiki.ros.org/tf#static_transform_publisher

james-yoo, 2022-12-27:
@chivas1000 
Thanks for your reply. Yes, I know that is a static transform from map frame to odom frame. What I like to know is the meaning of the physical value '-0.3'. Is it related to the position of camera sensor?

chivas1000, 2022-12-27:
@james-yoo 
I think this related to position of camera, because the odom frame should indicates where the camera is, and map is where the ground is.
and in the simulation, the robot isaac mounts the camera 0.3m off the ground.

james-yoo, 2023-01-03:
@chivas1000 
Great! Thanks for your kind response.

closed 2023-01-03 by james-yoo