NVIDIA-ISAAC-ROS/isaac_ros_nvblox #50: Mesh block is too big to weld. Exiting without doing anything.

chivas1000, 2022-12-27:
hi,

I’m using velodyne vlp16 lidar as nvblox input

at first it shows up correct mesh even if I move the robot for few meters
[image]

but after I move the robot further and steady, the robot in just teleported to a different area,
[image]

and then the mesh just mess up,

and terminal has output

[nvblox_node-10] WARNING! Mesh block is too big to weld. Exiting without doing anything.
[nvblox_node-10] WARNING! Mesh block is too big to weld. Exiting without doing anything.
[nvblox_node-10] WARNING! Mesh block is too big to weld. Exiting without doing anything.

I thought it might be the allocated block are not enough?

I have tuned the voxel_size larger to 0.1 so as to reduce computation, but it seems no use.

so how can I tune for it?

thanks for any assistance.

-----------------------------------------------------------------------------
nvblox_lidar.yaml:

---
nvblox_node:
  ros__parameters:
    global_frame: odom
    #voxel size changed from 0.05 to 0.1 to lower the computation
    voxel_size: 0.1
    slice_height: 0.75
    #min_height: -0.25
    #min_height: -0.1
    min_height: 0.1
    max_height: 1.0
    esdf: true
    esdf_2d: true
    mesh: true
    distance_slice: true
    use_color: true

    # lidar parameters:
    use_depth: false
    use_lidar: true
    lidar_width: 1800
    lidar_height: 16
    lidar_vertical_fov_rad: 0.5235987756

    tsdf_integrator_max_integration_distance_m: 4.0
    lidar_tsdf_integrator_max_integration_distance_m: 10.0
    tsdf_integrator_truncation_distance_vox: 4.0
    tsdf_integrator_max_weight: 20.0
    esdf_integrator_min_weight: 2.0
    esdf_integrator_max_distance_m: 2.0
    esdf_integrator_max_site_distance_vox: 5.0
    max_tsdf_update_hz: 0.0
    max_pointcloud_update_hz: 0.0
    max_color_update_hz: 5.0
    max_mesh_update_hz: 1.0
    max_esdf_update_hz: 10.0
    #lidar_width: 1800
    #lidar_height: 31
    output_dir: /tmp/

hemalshahNV, 2023-01-03:
Changes in `librealsense` and `realsense_ros` have caused some issues with Isaac ROS Nvblox. We've pushed out hotfixes here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/pull/49 that should help resolve this.

alexmillane, 2023-01-25:
So the message:

`WARNING! Mesh block is too big to weld. Exiting without doing anything`

Isn't actually a problem... It's misleading, but it just means that the mesh is not fully compressed, but it still valid.

Have you had any luck getting improved results? To me it looks like the pose is not well estimated (but it's hard to tell from an image).

closed 2023-02-14 by jaiveersinghNV