NVIDIA-ISAAC-ROS/isaac_ros_nvblox #51: How does nvblox node consumes topics of VSLAM node?

naitiknakrani, 2023-01-12:
Hi Team,
We are working with Jetson Orin and v0.11.0-dp Isaac ROS GEMS packages.

As per the following image shown, nvblox and vslam is linked via /pose topic (or data).
[image]

However if I checked ros2 node info /nvblox_node and
[image]

ros2 node info /visual_slam_node,
[image]

it looks like NVBLOX node is subscribed to /pose topic, however VSLAM node is publishing /pose topic. Also, we can not able to see REMAPPINGS of topics in any of the launch file given in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/tree/v0.11.0-dp/nvblox_nav2/launch or https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/tree/v0.11.0-dp/nvblox_examples/nvblox_examples_bringup/launch

How does than nvblox consuming data published by vslam node?

naitiknakrani, 2023-01-24:
Hi, Is there anyway Nvblox consumes /tf data ?

alexmillane, 2023-01-24:
You're correct, nvblox consumes the pose produced by VSLAM using the tf tree.

Hopefully that answers your question?

naitiknakrani-eic, 2023-01-27:
@alexmillane Thanks for the reply.

closed 2023-01-30 by hemalshahNV