NVIDIA-ISAAC-ROS/isaac_ros_nvblox #54: Nvbox in Isaac Gym (the Omniverse version)

joehays, 2023-03-03:
Hello,

Is there any support to be able to use the nvblox capabilities from Isaac ROS nvbox in parallel environments in Omniverse Sim/Gym? The ROS2 layer would have to be stripped away as it's not appropriate for the parallel Gym environments used during the RL training. Is this possible?

alexmillane, 2023-03-04:
I'm not super familiar with Isaac Gym but from what I can see you interact with it through pytorch tensors? I guess then to connect Isaac Gym with nvblox the missing piece is a pytorch interface to nvblox? Is that correct?

Internally, we have a pytorch interface, but it's not ready to be released. So for now you'll have to wrap the bits you need yourself. To do this you would wrap parts of the [nvblox core library](https://github.com/nvidia-isaac/nvblox).

joehays, 2023-03-05:
how about the ability to safely instantiate multiple nvblox objects in parallel (i.e. in parallel Gym environments)?

joehays, 2023-03-05:
Also, are there plans to make the pytorch interface available in the near future?