NVIDIA-ISAAC-ROS/isaac_ros_nvblox #55: rosdep install fails in nvblox_vslam_realsense tutorial

RezaOptimotive, 2023-03-09:
So I'm just trying to get nvblox working with realsense so I followed the steps [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-nvblox-vslam-realsense.md#installing-the-dependencies), twice.

There are two things that happen, 
1. when I run 
```
 cd /tmp && \
 git clone https://github.com/IntelRealSense/librealsense && \
 cd librealsense && \
 ./scripts/setup_udev_rules.sh
```

from [here](https://github.com/NVIDIA-ISAAC-ROS/.github/blob/main/profile/realsense-setup.md), after entering my password I get a lot of permissions denied. So I ran the script using sudo and it works this time.

2. Trying to do `rosdep installs` inside the container from the 5th step from [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-nvblox-vslam-realsense.md#installing-the-dependencies) I get this:

```
$ rosdep install -i -r --from-paths src --rosdistro humble -y --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv nvblox"
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: No definition of [librealsense2] for OS version [focal]
realsense2_description: No definition of [xacro] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

RezaOptimotive, 2023-03-09:
So I updated the .isaac_ros_common-config file for x86:
```
cat scripts/.isaac_ros_common-config                                                                          
CONFIG_IMAGE_KEY=x86_64.humble.nav2.realsense
```

but the issue persists unfortunately

batzor, 2023-04-06:
Did you resolve it? I am stuck with the same issue.

ARK3r, 2023-04-10:
I've had to move to other projects before I got the chance to get it working, so unfortunately no.

jaiveersinghNV, 2023-04-10:
We've updated this package and the accompanying documentation in our new [Isaac ROS DP3 Release](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/releases/tag/v0.30.0-dp).

Could you please try again using the latest release, and let us know if the issue persists?

ARK3r, 2023-04-10:
I will test it out and let you know.

ARK3r, 2023-04-10:
The demo works and it is magnificent. 

Next, I tried to run the realsense example and that one seems to fail. It appears that the realsense wrapper isn't even installed. I'll try to post an update here once I figure more of it out.

ARK3r, 2023-04-11:
The refurbished packages resolved my issue here, so I think we can close this.

I'll need to open another one regarding dkms and kernel versions though :worried:

closed 2023-04-13 by gordongrigor