NVIDIA-ISAAC-ROS/isaac_ros_nvblox #62: colcon build --symlink-install  error

kong9999, 2023-05-05:
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-human-reconstruction-realsense.md
Hello, I ran into this problem when executing this example.
[image]
Starting >>> isaac_ros_stereo_image_proc
Finished <<< isaac_ros_nitros_detection2_d_array_type [26.0s]  
--- stderr: isaac_ros_image_proc                               
CMake Error at gxf/image_flip/CMakeLists.txt:21 (include):
  include could not find requested file:

    YamlCpp

CMake Error at gxf/image_flip/CMakeLists.txt:22 (find_package):
  No "FindGXF.cmake" found in CMAKE_MODULE_PATH.

CMake Warning (dev) at gxf/image_flip/CMakeLists.txt:22 (find_package):
  FindGXF.cmake must either be part of this project itself, in this case
  adjust CMAKE_MODULE_PATH so that it points to the correct location inside
  its source tree.

  Or it must be installed by a package which has already been found via
  find_package().  In this case make sure that package has indeed been found
  and adjust CMAKE_MODULE_PATH to contain the location where that package has
  installed FindGXF.cmake.  This must be a location provided by that package.
  This error in general means that the buildsystem of this project is relying
  on a Find-module without ensuring that it is actually available.

This warning is for project developers.  Use -Wno-dev to suppress it.

---
Failed   <<< isaac_ros_image_proc [2.16s, exited with code 1]
Aborted  <<< isaac_ros_stereo_image_proc [2.80s]                          
Aborted  <<< isaac_ros_visual_slam [27.6s]                                
Aborted  <<< nvblox [4min 15s]

hemalshahNV, 2023-05-08:
The latest Isaac ROS Common is needed with updated packages with these additional cmake files. Could you also check whether you have a clean build of your ROS workspace (rm -Rf build install log) to make sure you pick up the latest?

BloodLemonS, 2023-05-17:
Have you encountered the following problems and how to solve them?
[image]

jaiveersinghNV, 2023-05-17:
This might be a transient network issue somewhere along the connection between your machine and GitHub's servers. I'm able to successfully clone [the repository](https://github.com/stotko/stdgpu.git) on my machine.

Please double-check that your machine can access GitHub/the internet from within the container.

Additionally, please create a separate GitHub issue next time. There are lots of potential causes for the build to fail, and separating the issues out helps us track and respond more effectively, while also providing more value for future readers 😃

closed 2023-06-05 by jaiveersinghNV