NVIDIA-ISAAC-ROS/isaac_ros_nvblox #64: Carter can’t move in the narrow area.

mat-macnica, 2023-05-16:
Hello,

We can succeeded to run below demo.
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-human-reconstruction-isaac-sim.md

But Robot(Carter) sometimes go into narrow area.
Then the Carter can't move like below;
[image]

Question:
 Are there any recommended way to solve this problem?

--
Best regards,

jaiveersinghNV, 2023-05-31:
You may have to tune your costmap parameters to adjust how much the boundary around obstacles is inflated. In your scenario, it looks like the robot size is larger than the available freespace between the two obstacles, causing the robot to get stuck. You might also modify the parameters for the motion planner to ensure that the entire path is clear before proceeding between obstacles.

mat-macnica, 2023-06-01:
Hello,

Thank you very much for your kindness.
I really appreciate your help.

I'd like to check the behavior with your guide.

Question:　
　- Could you tell us how to change  the costmap parameters?
  
We would appreciate if you tell us which file (xx.py?)  should be modified to tune the costmap parameters of below demo.
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-human-reconstruction-isaac-sim.md

--
Best regards,

alexmillane, 2023-06-03:
Hi there. So the parameters which you need to play around with are the nav2 parameters. In particular the parameters you should look at are those related to the nvblox costmap plugin. One easy thing you could try is to reduce the robot radius here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/nvblox_examples/nvblox_examples_bringup/config/nav2/nav2_isaac_sim.yaml#L178

Other parameters in that file could also be worth trying, for example the obstacle inflation distance: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/nvblox_examples/nvblox_examples_bringup/config/nav2/nav2_isaac_sim.yaml#L178

closed 2023-07-26 by jaiveersinghNV