NVIDIA-ISAAC-ROS/isaac_ros_nvblox #65: Nvblox with Realsense - Odom Frame does not exist / DKMS Kernel version

ChrwagnerTHU, 2023-05-16:
Hi,

I'm currently trying to run nvblox on my x86 system with a realsense.
As with some other issues here, I'm facing the following error `Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist`
I followed [this](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/troubleshooting-nvblox-realsense.md#solution-1---install-dkms) guide in order to install librealsense2-dkms. 
Unfortunately, this guide does not work for me. My system runs on kernel version 5.19.0-41-generic. The linked download at the guide only supports 5.13/5.15.
As far as I see it, there is no librealsense2-dkms package for kernel version 5.19.

Is there a possible solution to solve this issue without downgrading my kernel?

Many thanks in advance!

ChrwagnerTHU, 2023-05-24:
I've downgraded my kernel, and installed the package from source. After that nvblox worked once, but after that never again.
I think it's important to mention that I work on the NVIDIA Isaac ROS Docker. So every installation of packages is not persistent.
Even if I redo everything to install the package again, it does not work.
Is there a way of making the installation of the new package persistent?

jaiveersinghNV, 2023-05-31:
We provide a [Dockerfile](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.realsense) that performs the necessary setup and installation for RealSense while building the container, which saves you from having to install it each time. To use it, create a file in `isaac_ros_common/docker/scripts` named `.isaac_ros_common-config` that specifies `CONFIG_IMAGE_KEY=ros2_humble.realsense`, as explained [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docs/modify-dockerfile.md#change-the-config-image-key).

closed 2023-06-01 by ChrwagnerTHU