NVIDIA-ISAAC-ROS/isaac_ros_nvblox #66: Failed to build base image: isaac_ros_dev-x86_64, aborting

BloodLemonS, 2023-05-17:
What should I do when I get the following error while running./scripts/run_dev.sh?

failed to copy: httpReadSeeker: failed open: failed to authorize: rpc error: code = Unknown desc = failed to fetch anonymous token: unexpected status: 401 Unauthorized

jaiveersinghNV, 2023-05-17:
It sounds like you may need to create an NGC account and properly authorize, as explained [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/82#issuecomment-1520944262). 

We're working on a fix that will restore anonymous layer fetching and building.

dev10110, 2023-06-27:
Ive noticed that sometimes restarting the docker daemon helps fix issues (might not fix this particular one)

```sudo systemctl restart docker```

closed 2023-07-26 by swapnesh-wani-nvidia