NVIDIA-ISAAC-ROS/isaac_ros_nvblox #70: `Failed to call git rev-parse` and `Failed to build base image: isaac_ros_dev-x86_64, aborting.`

dev10110, 2023-06-22:
Hi, 

Ive been trying to build the nvblox library, and Ive followed instructions here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-realsense.md but I get stuck at the following:

```
cd ~/workspaces/isaac_ros-dev/src/isaac_ros_common && \
>   ./scripts/run_dev.sh
```
has output
```
isaac_ros_dev not specified, assuming /home/dasc3/workspaces/isaac_ros-dev
Error: Failed to call git rev-parse --git-dir: exit status 128 
Building x86_64.ros2_humble.realsense.user base as image: isaac_ros_dev-x86_64 using key x86_64.ros2_humble.realsense.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Failed to build base image: isaac_ros_dev-x86_64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common
```

any suggestions will be helpful!

Thanks
Dev

jaiveersinghNV, 2023-06-26:
Could you try the following steps:
1. In `isaac_ros_common/scripts/build_base_image.sh`, replace line 1 to read:
`#!/bin/bash -x`
This will ensure that each command is printed out, to help us understand exactly where the script is failing.
2. On your host machine, run `nvidia-smi`. Sometimes, this can fail because of incorrect GPU driver versions on your host machine, which will then cause the container build process to fail.

dev10110, 2023-06-27:
Hi, I think I was just using not using the `run_dev.sh` script correctly. Providing the workspace path seems have fixed the issue. 

Thanks!

closed 2023-06-27 by dev10110