NVIDIA-ISAAC-ROS/isaac_ros_nvblox #72: Occupancy grid not cleared if map_clearing_radius_m > 0 and use_static_occupancy_layer: true

shtern94, 2023-06-28:
Hello,
Is there a way to clear occupancy grid based on radius? I trying adjusting parameters, but it seems that map clearing is supported only in tsdf mode (e.g. use_static_occupancy_layer: false).
Also is it possible to clear the occupancy grid based on decay threshold? If not, can this feature can be added in future?
Thank you in advance

shtern, 2023-07-13:
@jaiveersinghNV thank you for assigning the team member to this issue, I've discovered there's an uncovered scenario in the nvblox repo. I've issued the PR [#33](https://github.com/nvidia-isaac/nvblox/pull/33

shtern, 2023-07-17:
@alexmillane so half of the question was fixed in the PR, thanks for approving it. Can you please answer about the decay based filtering? E.g. cleaning out the map after parts of the grid weren't seen for the long period of time?

alexmillane, 2023-07-25:
Hi @shtern . Thank you for your interest in nvblox.

So we actually have the tools in nvblox for decaying occupancy values. We use it in the human mapper to decay human occupancy voxels: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/nvblox_ros/include/nvblox_ros/nvblox_human_node.hpp#L81

It would be relatively straightforward to add an option to decay the non-human occupancy layer inside the normal nvblox node.

If you add that, I'm happy to review an MR :)

closed 2023-08-21 by jaiveersinghNV