NVIDIA-ISAAC-ROS/isaac_ros_nvblox #78: How could the robot localize after loading a saved map?

daodao9801, 2023-10-11:
I am using the realsense D435i with the given example successfully.

`ros2 launch nvblox_examples_bringup realsense_example.launch.py`

After loading my saved map

`ros2 service call /nvblox_node/load_map nvblox_msgs/srv/FilePath "{file_path: '/home/USERNAME/super_cool_map.nvblx'}"`

The map can be seen together with "Esdf Slice" afterwards and everything looks fine. However, there seems no localization happening when the new map is loaded. 

May I know how to let the robot find its position when a map is loaded? Is it automatically done or should I give it a "2D pose estimation" via RViz? Is it failed because the current depth map doesnt have enough feature?

Thanks advance for any suggestions.

alexmillane, 2023-10-21:
nvblox, on its own, is just a mapping pipeline. Mapping being half of Simultaneous Localization and Mapping (SLAM).

In other words, nvblox on its own will not relocalize in a map. We expect another module to do that. You _could_ configure, for example, isaac_ros_visual_slam, or another visual slam system, to do the relocalization.

daodao9801, 2023-10-26:
Thanks. So `ros2 action send_goal /visual_slam/save_map isaac_ros_visual_slam_interfaces/action/SaveMap "{map_url: /map}` would be a more appropriate choice for saving a map which could be used for localizing later in the isaac_ros_visual_slam process?

jaiveersinghNV, 2023-10-30:
Yes, that would be an appropriate usage of the [`save_map` and `load_map_and_localize` actions](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_visual_slam/isaac_ros_visual_slam/index.html#ros-actions-advertised) in Isaac ROS Visual SLAM.

closed 2023-11-21 by jaiveersinghNV