NVIDIA-ISAAC-ROS/isaac_ros_nvblox #79: Two camera integration pipeline

naitiknakrani-eic, 2023-10-12:
Hi Team, 

I am interested to know whether the present pipeline running with VSLAM and NVBLOX can be modified into the following figure?

Here, I want to use depth of ToF camera and Infra1 and Infra2 of Realsense. I want to reconstruct 3D environment with ToF instead of Realsense. Note: I am putting both camera on stack. 

[image]

Are there any limitations or logical reasoning that it shouldn't work ?

naitiknakrani-eic, 2023-10-16:
Hi, 

Please find the attached image. I tried to feed depth data from ToF into NVBLOX with wheel odometry as a input. Typically with Realsense it works (with only depth data). 

Is there any recommendation to resolve the working of NVBLOX reconstruction using ToF's depth data?

[image]

Right side shows the logs of NVBLOX and left side shows the depth data feed.

alexmillane, 2023-10-21:
The usual reason for getting no reconstruction is:
- A topic mismatch.
- A missing link between the sensor frame and the mapping frame on the TF tree.

Could you check if the depth topic is correctly mapped such that depth images are arriving at nvblox? Then I would check if a link exists between your depth optical frame (which is specified in the `frame_id` entry in your depth image header), and the mapping frame, which is by default `odom`.

naitiknakrani-eic, 2023-10-25:
Hi @alexmillane Thank you for your quick response and apologies for not giving enough info. 

I am sharing topics name, topics info, changes done in files in this post.

First, my camera topics are:
> /cam1/depth_image_nvblox
> /cam1/camera_info_sync
and my camera frame id is > "camera_link_optical"

 **Topics mismatch**
I have remapped two topics in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/v2.0.0/nvblox_examples/nvblox_examples_bringup/launch/nvblox/nvblox.launch.py
[image]

and changed frame id in config file given at https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/v2.0.0/nvblox_examples/nvblox_examples_bringup/config/nvblox/specializations/nvblox_realsense.yaml
[image]

I have checked weather the data is arriving and proper? It seems perfect as shown in below image.
[image]
[image]

**A missing link between the sensor frame and the mapping frame on the TF tree.**

Once I launched the node, I checked TF tree. Here is the image showing my TF tree.
[image]

Let me know If I missed to provide any information.

naitiknakrani-eic, 2023-10-25:
Note that I have tested it in both v.2.0 and v0.31.0-dp.

naitiknakrani-eic, 2023-11-03:
Hi @alexmillane any inputs/suggestions ?

ashwinvkNV, 2023-11-10:
Hi, @naitiknakrani-eic can you post the following so we can try to reproduce this?
Rosbag containing:
1. real sense stereoimages
2. Tof images
3. Tf
Launch file
Nvblox config file

closed 2024-01-09 by jaiveersinghNV

naitiknakrani-eic, 2024-01-09:
Thank you @alexmillane and @ashwinvkNV for your support. The time sync between depth and camera_info works for us and nvblox gave voxel map.