NVIDIA-ISAAC-ROS/isaac_ros_nvblox #85: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'condition'

SunilSrivatsav9, 2023-11-29:
The work seems to be too interesting. I have downloaded the repository few months back and tested the isaac_ros_nvblox. I tested the tutorials in isaac sim simulator as well as using real sense camera for real-time. Everything worked well. 
However there is new update in the package and dynamic tutorials were added. So I downloaded the isaac_ros_nvblox package and replaced with old package and build again for checking the tutorials. However I have issue with condition statement as below. Can you help me out in solving the issue. Is it because of the mismatch ros version or any other dependency issue.

Below is the log fie
```
1701221156.1139338 [INFO] [launch]: Default logging verbosity is set to INFO
1701221156.3357425 [ERROR] [launch]: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'condition'
1701221156.5786924 [INFO] [component_container_mt-1]: process started with pid [29559]
1701221156.5797582 [INFO] [component_container_mt-1]: sending signal 'SIGINT' to process[component_container_mt-1]
1701221156.6851957 [INFO] [component_container_mt-1]: process has finished cleanly [pid 29559]
1701221157.5612414 [WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'shared_nvblox_container/_container/load_node' service response, due to shutdown.
```

This is the complete error 
```
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'condition'
[INFO] [component_container_mt-1]: process started with pid [720]
[component_container_mt-1] [INFO] [1701222822.699096851] [shared_nvblox_container]: Load Library: /workspaces/isaac_ros-dev/install/realsense_splitter/lib/librealsense_splitter_component.so
[component_container_mt-1] [INFO] [1701222822.704229477] [shared_nvblox_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container_mt-1] [INFO] [1701222822.704279312] [shared_nvblox_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvblox::RealsenseSplitterNode>
[component_container_mt-1] [INFO] [1701222822.709484685] [camera.realsense_splitter_node]: Creating a RealsenseSplitterNode().
[INFO] [component_container_mt-1]: sending signal 'SIGINT' to process[component_container_mt-1]
[component_container_mt-1] [INFO] [1701222822.718272949] [rclcpp]: signal_handler(signum=2)
[INFO] [component_container_mt-1]: process has finished cleanly [pid 720]
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the 'shared_nvblox_container/_container/load_node' service response, due to shutdown.
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/install/lib/python3.8/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/install/lib/python3.8/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/install/lib/python3.8/site-packages/rclpy/client.py", line 180, in wait_for_service
    return self.service_is_ready()
  File "/opt/ros/humble/install/lib/python3.8/site-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
```

jaiveersinghNV, 2023-12-04:
What version of ROS are you using? Can you confirm that your system matches the specific [device requirements](https://nvidia-isaac-ros.github.io/getting_started/index.html#system-requirements) for Isaac ROS?

[This thread](https://github.com/ros-planning/navigation2/issues/3840#issuecomment-1737827402) suggests that you might be on a too-new version of ROS. Note that Isaac ROS currently targets ROS 2 Humble, not Iron.

KitsuELP, 2024-01-04:
I'm having the same issue trying to run the realsense example. Im using the Jetson Orin Nano using the prebuilt ROS Isaac SD image. My camera is the D435 (no IMU). Any updates to this issue?

jaiveersinghNV, 2024-01-09:
Hi @KitsuELP ,

Are you referring to the SD card image provided [here](https://nvidia-ai-iot.github.io/jetson_isaac_ros_visual_slam_tutorial/sdcard.html)?

If so, the Isaac ROS software pre-installed on the SD card is from the older DP3 release. When you download any additional Isaac ROS repositories like `isaac_ros_nvblox`, you should make sure to get the compatible [DP3 release version](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/releases/tag/v0.30.0-dp), *not* the latest 2.1 release version.

Could you please confirm that you have all Isaac ROS packages on a consistent version and then try again?