NVIDIA-ISAAC-ROS/isaac_ros_nvblox #89: isaac sim example crashes

ARK3r, 2024-01-18:
trying to run the [simple isaac_ros_nvblox isaac sim example](https://nvidia-isaac-ros.github.io/concepts/scene_reconstruction/nvblox/tutorials/tutorial_isaac_sim.html) when running `./python.sh ${ISAAC_ROS_WS}src/isaac_ros_nvblox/nvblox_examples/nvblox_isaac_sim/omniverse_scripts/start_isaac_sim.py --gpu_physics_enabled` from inside `~/.local/share/ov/pkg/isaac_sim-2023.1.1` a lot of logs are printed and isaac sim program starts, then after a while some models are shown, then immediately the program crashes with the following output:

```
[6.045s] Simulation App Starting
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2024-01-17 19:52:03 [6,053ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-01-17 19:52:04 [6,957ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-01-17 19:52:04 [6,957ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[7.769s] app ready
2024-01-17 19:52:05 [8,090ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-01-17 19:52:05 [8,090ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.225s] RTX ready
[8.446s] Simulation App Startup Complete
[8.455s] [ext: omni.isaac.ros2_bridge-2.5.1] startup
[8.485s] Attempting to load system rclpy
[8.506s] rclpy loaded
Opening stage omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Samples/NvBlox/carter_warehouse_navigation.usd...
2024-01-17 19:53:47 [109,715ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </World/Carter_ROS>: Could not open asset @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Robots/Carter/carter_v2.usd@ for reference introduced by @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Samples/NvBlox/Robots/Carter_v2_ROS2.usd@</Carter>. (instantiating stage on stage @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Samples/NvBlox/carter_warehouse_navigation.usd@ <0xd094880>)

2024-01-17 19:53:47 [109,715ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </World/Carter_ROS>: Could not open asset @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Robots/Carter/carter_v2.usd@ for reference introduced by @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Samples/NvBlox/Robots/Carter_v2_ROS2.usd@</Carter>. (instantiating stage on stage @omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.1/Isaac/Samples/NvBlox/carter_warehouse_navigation.usd@ <0xd094880>)

2024-01-17 19:53:47 [109,735ms] [Warning] [omni.timeline.plugin] Deprecated: direct use of ITimeline callbacks is deprecated. Use ITimeline::getTimeline (Python: omni.timeline.get_timeline_interface) instead.
2024-01-17 19:53:47 [109,875ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x22dd29c0 usdValid=0 cpuValid=0
Loading stage...
Loading Complete
Running sim at 20 Hz, with dt of 0.05
2024-01-17 19:53:53 [115,748ms] [Warning] [carb] Client omni.physx.plugin has acquired [omni::physics::schema::IUsdPhysics v1.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2024-01-17 19:53:53 [115,944ms] [Warning] [omni.isaac.dynamic_control.plugin] Failed to find articulation at '/World/Carter_ROS/chassis_link'
2024-01-17 19:53:53 [115,944ms] [Error] [omni.isaac.dynamic_control.plugin] DcGetArticulationRootBody: Invalid or expired articulation handle
2024-01-17 19:53:53 [115,944ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/Carter_ROS/chassis_link'
2024-01-17 19:53:53 [115,944ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/Carter_ROS/chassis_link' did not match any articulations

2024-01-17 19:53:53 [116,129ms] [Warning] [omni.graph.core.plugin] /World/Carter_ROS/ActionGraph/articulation_controller: [/World/Carter_ROS/ActionGraph] OmniGraph Warning: 'NoneType' object has no attribute 'is_homogeneous'
                   (from compute() at line 112 in /home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core_nodes/omni/isaac/core_nodes/ogn/python/nodes/OgnIsaacArticulationController.py)
2024-01-17 19:53:53 [116,134ms] [Error] [omni.graph.core.plugin] /World/Carter_ROS/ActionGraph/isaac_read_lidar_point_cloud_node: [/World/Carter_ROS/ActionGraph] Prim is not a Lidar Prim
2024-01-17 19:53:53 [116,136ms] [Error] [omni.graph.core.plugin] /World/Carter_ROS/ActionGraph/isaac_compute_odometry_node: [/World/Carter_ROS/ActionGraph] prim is not a valid rigid body or articulation root
2024-01-17 19:53:53 [116,436ms] [Warning] [omni.isaac.dynamic_control.plugin] Failed to find articulation at '/World/Carter_ROS/chassis_link'
2024-01-17 19:53:53 [116,436ms] [Error] [omni.isaac.dynamic_control.plugin] DcGetArticulationRootBody: Invalid or expired articulation handle
2024-01-17 19:53:53 [116,436ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/Carter_ROS/chassis_link'
2024-01-17 19:53:53 [116,436ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/Carter_ROS/chassis_link' did not match any articulations

terminate called without an active exception
Fatal Python error: Aborted

Thread 0x00007f0edcfff640 (most recent call first):
  File "/home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007f132bb1bb80 (most recent call first):
  File "/home/reza/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 701 in step
  File "/home/reza/workspaces/isaac_ros-dev//src/isaac_ros_nvblox/nvblox_examples/nvblox_isaac_sim/omniverse_scripts/start_isaac_sim.py", line 270 in main
  File "/home/reza/workspaces/isaac_ros-dev//src/isaac_ros_nvblox/nvblox_examples/nvblox_isaac_sim/omniverse_scripts/start_isaac_sim.py", line 371 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix, pydantic.version, pydantic.datetime_parse, pydantic.validators, pydantic.networks, pydantic.schema, pydantic.main, pydantic.env_settings, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, omni.mdl.pymdlsdk._pymdlsdk, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, [... cut ...]
./python.sh: line 41: 32551 Aborted                 (core dumped) $python_exe "$@" $args
There was an error running python
```

I'm trying to run isaac sim natively on an ubuntu 22.04. I just updated isaac sim and cache and followed the instructions [here](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_workstation.html) to setup cache and nucleus and server. I also have nvidia driver 525. The isaac_ros_nvblox package is also up to date.

Any suggestions as to what I should do here?

remostei, 2024-02-06:
Is this still an issue @ARK3r? 
We updated the sim assets to fix it.

ARK3r, 2024-02-07:
yes it works, thanks!

closed 2024-02-07 by ARK3r

djsamseng, 2024-02-14:
How can we get the updated sim assets if we downloaded them as the Isaac Sim Asset Packs? Issac Sim Asset Pack 1 only has carter_v1.usd and not carter_v2.usd. Due to being abroad, downloading assets through Nucleus is either very slow or fails to start the download. Thanks!