NVIDIA-ISAAC-ROS/isaac_ros_nvblox #90: Realsense Firmware for vslam and nvblox

ikhann, 2024-01-22:
Hi!

I am planning to use isaac_ros_visual_slam in conjunction with isaac_ros_nvblox, and I'm using the Realsense D435 camera for this setup. However, I've encountered a bit of a dilemma regarding the firmware version compatibility between these two ROS packages and the Realsense camera.

The [isaac_ros_visual_slam Troubleshooting Suggestions](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_visual_slam/isaac_ros_visual_slam/index.html#troubleshooting-suggestions) advise that the firmware of the RealSense camera should be [5.14.0 or newer](https://dev.intelrealsense.com/docs/firmware-releases). This seems straightforward, but there's a conflicting recommendation for isaac_ros_nvblox.

For isaac_ros_nvblox, the documentation indicates it has been tested and found compatible with RealSense camera firmware [version 5.13.0.50](https://dev.intelrealsense.com/docs/firmware-releases). More so, there's a specific note in the [nvblox examples](https://nvidia-isaac-ros.github.io/concepts/scene_reconstruction/nvblox/tutorials/tutorial_realsense.html#static-reconstruction-with-realsense) stating, "_The latest RealSense firmware (version 5.14 at the time of publishing) might have issues. Our recommendation is to install exactly_ [version 5.13.0.50"](https://dev.intelrealsense.com/docs/firmware-releases).

Given these conflicting recommendations, I'm unsure which firmware version to use for optimal performance and compatibility with both isaac_ros_visual_slam and isaac_ros_nvblox. 
Could you provide guidance on which firmware version would be the best choice in this situation?

Thank you!

jaiveersinghNV, 2024-01-29:
Hi @ikhann  - I believe we've responded to your [Developer Forum thread](https://forums.developer.nvidia.com/t/realsense-firmware-for-isaac-ros-visual-slam-and-isaac-ros-nvblox/279769) already, but for the sake of closure on this issue, I'll duplicate our response here:
> We suggest to use [version 5.13.0.50](https://dev.intelrealsense.com/docs/firmware-releases)

ikhann, 2024-01-29:
> Hi @ikhann  - I believe we've responded to your [Developer Forum thread](https://forums.developer.nvidia.com/t/realsense-firmware-for-isaac-ros-visual-slam-and-isaac-ros-nvblox/279769) already, but [...]

Indeed! Thank you for your help!