NVIDIA-ISAAC-ROS/isaac_ros_nvblox #93: Process dies after /nvblox_node/save_ply service call

IABSTUDIO, 2024-01-26:
Hello,

I'm running isaac_ros docker with NVIDIA Jetson Orin AGX and ZED X mini camera, and I'm trying to save real world's mesh file with `zed_example.launch.py`.

A few seconds after calling /save_ply service as `ros2 service call /nvblox_node/save_ply nvblox_msgs/srv/FilePath "{file_path: '/home/admin/super_cool_map.ply'}"`, camera and Rviz stopped working, and launch file's process died.

Only error code I could see in terminal was this:
`[ERROR] [component_container_mt-1]: process has died [pid 19410, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=shared_nvblox_container'].`

Could you tell me how to fix this?

IABSTUDIO, 2024-01-30:
I found that process dies while running 'for' loop of `Mesh::fromLayer` function in `nvblox/mesh/mesh.cpp` if camera is running.
When I unplugged camera before calling `save_ply` service, process didn't die and could save .ply file successfully.

But still don't know how to fix this problem:(

Aeyu17, 2024-07-25:
Hello, I've also had this issue on Isaac ROS 3.0 and have been able to replicate unplugging the camera. If there are any updates to this issue please let me know!