NVIDIA-ISAAC-ROS/isaac_ros_nvblox #98: colcon build --symlink-install failed (Native Installation Ros2)

asadwadood, 2024-03-06:
Hello,
I am facing an error while native installation. I am running on a PC having ubuntu 20, ROS2 Foxy. 

Finished <<< nvblox_isaac_sim [9.00s]
Starting >>> nvblox_image_padding
Finished <<< nvblox_examples_bringup [10.8s]                                                                                                                                                  
Finished <<< nvblox [14.1s]                                                                                                                                                                                       
Finished <<< odometry_flattener [10.6s]                                                                                                                                                   
Finished <<< nvblox_msgs [16.9s]                                                                                                                     
Finished <<< nvblox_performance_measurement_msgs [14.9s]                                                               
Starting >>> nvblox_ros
Starting >>> nvblox_nav2                                                                   
Starting >>> nvblox_rviz_plugin                                                                                 
Starting >>> semantic_label_conversion                                                                                                      
Finished <<< nvblox_image_padding [8.48s]                                                                                                                                         
Finished <<< semantic_label_conversion [3.06s]                                                                                                                                
Finished <<< nvblox_nav2 [6.23s]                                                                                                   
Finished <<< nvblox_rviz_plugin [6.25s]                                                               
--- stderr: nvblox_ros                                 
In file included from /home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/mapper_initialization.hpp:28,
                 from /home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/src/lib/mapper_initialization.cpp:18:
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp: In function ‘void nvblox::declareParameterWithoutDefault(const string&, rclcpp::Node*)’:
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:36:78: error: expected unqualified-id before ‘&’ token
     const rclcpp::exceptions::UninitializedStaticallyTypedParameterException & ex)
                                                                              ^
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:36:77: error: expected ‘)’ before ‘&’ token
     const rclcpp::exceptions::UninitializedStaticallyTypedParameterException & ex)
                                                                             ^~
                                                                             )
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:35:11: note: to match this ‘(’
   } catch (
           ^
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:36:78: error: expected ‘{’ before ‘&’ token
     const rclcpp::exceptions::UninitializedStaticallyTypedParameterException & ex)
                                                                              ^
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:36:80: error: ‘ex’ was not declared in this scope
     const rclcpp::exceptions::UninitializedStaticallyTypedParameterException & ex)
                                                                                ^~
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:36:80: note: suggested alternative: ‘exp’
     const rclcpp::exceptions::UninitializedStaticallyTypedParameterException & ex)
                                                                                ^~
                                                                                exp
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp: In instantiation of ‘void nvblox::declareParameterWithoutDefault(const string&, rclcpp::Node*) [with ParameterType = bool; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/src/lib/mapper_initialization.cpp:93:50:   required from here
/home/a/ros_ws/src/isaac_ros_nvblox/nvblox_ros/include/nvblox_ros/utils.hpp:34:5: error: no matching function for call to ‘rclcpp::Node::declare_parameter<bool>(const string&)’
     node_ptr->declare_parameter<ParameterType>(name);
     ^~~~~~~~

Can you give me an advice? Thank you. 
[image]

ashwinvkNV, 2024-03-06:
Hi, unfortunately we currently only support ROS2 humble. And the build is **not** expected succeed with ROS 2 Foxy.

You can modify isaac_ros_nvblox to work around this build error, but you may face more issues.

Also in ROS 2 foxy, you will not be able to use NITROS since it depends on type_adaptation that was introduced in ROS 2 humble.
https://ros.org/reps/rep-2007.html
https://discourse.ros.org/t/new-type-adaptation-open-source-examples-for-humble/26241

asadwadood, 2024-03-07:
Thank you.  I successfully build it by using ROS2 humble

closed 2024-03-07 by asadwadood