NVIDIA-ISAAC-ROS/isaac_ros_nvblox #99: realsense_splitter error

davidlmljll, 2024-03-19:
I found that the output image of the splitter still has projection interference, but I have turned on emitter_on_off and the original camera also outputs correctly (shown as projection point images appearing at intervals)
[image]
At the same time, when I start nvblox, an error related to opencv will appear, as shown below. How to solve it?

HappySamuel, 2024-03-20:
I also faced the similar issue. At first, i follow the instructions here [https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_nvblox/isaac_ros_nvblox/troubleshooting/troubleshooting_nvblox_realsense.html](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_nvblox/isaac_ros_nvblox/troubleshooting/troubleshooting_nvblox_realsense.html) and do the setting.
Then i have the emitter setting as below. However, through rviz, discover that the splitter output image have emitter projection on it, which lead to visual slam cannot give a stable odometry.
```
depth_module:
  profile: '640x480x60'
  emitter_enabled: 1
  emitter_on_off: true
```

Any idea why? and how to solve this issue?

HariKrishnan06082k, 2024-11-14:
@davidlmljll  For your issue it looks like there's some incompatibility with the librealsense version and ros2 wrapper. I would also suggest you to build with the option of RSUSB_BACKEND true to avoid any incompatibility caused by kernel modules inside your docker (make sure it has --privileged flag as well). 

@HappySamuel I noticed this issue too apparently doing this as part of config file causes it to not work properly but after bringing up the camera node with the ros2 params **depth_module.emitter_enabled:0** and **depth_module.emitter_on_off: false** initially and once the realsense node is up and running in another terminal running ros2 param set /camera/camera **depth_module.emitter_enabled:0** followed by **depth_module.emitter_on_off: true** seems to work in the sense you dont see dotted IR projections from realsense splitter node. Idk why we can't trigger it at start in Launch Configuration level but it looks like a common problem which I'm facing too. you can time it set these params once the realsense camera node is up and running or have a sequential launch config which triggers once the realsense node camera is running. Unnecessary headache but yeah :(