NVIDIA-ISAAC-ROS/isaac_ros_nvblox #108: how can I use multi-cam option in Isaac ROS 3.0.0 May 30, 2024  patch note?

KimDoYoung1997, 2024-06-11:
Hello, Everyone..

I saw a new patch in Nvblox as below.
https://nvidia-isaac-ros.github.io/releases/index.html#isaac-ros-3-0-0-may-30-2024

As I understand, new **multi-cam** scene reconstruction and new multi-cam visual odometry is available according to new patch note..

But I can't find how to use multi-cam option in this repository.
Is there any specific document or manual that I can easily access? or can you guid me please??

Thank you!!!

alexmillane, 2024-06-11:
Hi there @KimDoYoung1997 

The `NvbloxNode` within nvblox has been upgraded to support multi-cam scene reconstruction.

At the moment we only have launchfiles which support this behaviour for [Perceptor](https://nvidia-isaac-ros.github.io/reference_workflows/isaac_perceptor/index.html). The best way to get started with the multi-cam behavior is to check out the [perceptor docs](https://nvidia-isaac-ros.github.io/reference_workflows/isaac_perceptor/index.html). For example, there's a tutorial on [running on pre-recorded data collected on a NOVA carter](https://nvidia-isaac-ros.github.io/reference_workflows/isaac_perceptor/tutorials_on_devkit/tutorial_recording_and_playback.html).

KimDoYoung1997, 2024-06-13:
> Hi there @KimDoYoung1997
>
> The `NvbloxNode` within nvblox has been upgraded to support multi-cam scene reconstruction.
>
> At the moment we only have launchfiles which support this behaviour for [...]

@alexmillane Thank you for reply! I will try it with Nova first. 
Someday you have any plan to add multi-cam option on Nvblox??

alexmillane, 2024-06-27:
So nvblox it's self _does_ support multi-cam right now. Currently, the only camera that we supply launchfiles for is the [Hawk camera](https://leopardimaging.com/leopard-imaging-hawk-stereo-camera/), which is part of the [NOVA sensor suite](https://robotics.segway.com/nova-dev-kit/). In principle, nothing prevents using any other camera, you need only supply the image to the relevant topics. The challenge, however, is getting multiple, time-synchronized image streams into ROS. We've solved that with NOVA, with significant effort. It's not-impossible that we add support for other cameras in the future.

closed 2024-06-28 by KimDoYoung1997