NVIDIA-ISAAC-ROS/isaac_ros_nvblox #110: nvblox with a different depth-sensing camera than RealSense

nhathout, 2024-06-13:
I am currently trying to use nvblox, specifically an altered version of the realsense example, using an ORBBEC Gemini 2 L camera instead. I have remapped topics that nvblox uses accordingly, as well as added the camera to "NvbloxCamera enum", but still seem to be getting errors. Is there any way someone could provide the files or locations where edits are necessary to adapt nvblox to work with another camera? The reason I am also asking this is because the more time I spend digging and trying to get this to work, the more directories I find with same (or very similar) names, so I am just never sure if I am editing enough to make this compatible and work (or if it will even work). 

At this point I don't care too much about quality issues, I would just like to see the nodes connected properly in rqt.

alexmillane, 2024-06-27:
I've not used the Orbec camera, but for the purposes of this answer, I'll assume that it's a 1:1 drop-in for the Realsense.

To make this work with the minimal changes, I would replace the realsense driver [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/nvblox_examples/nvblox_examples_bringup/launch/realsense_example.launch.py#L55) with the Orbec driver, and then remap the topics to match the realsense outputs exactly. In this configuration, nvblox will think it's operating from a realsense camera, but will be using the Orbec. If the assumption holds that the Orbec is a 1:1 replacement, this should work.

nhathout, 2024-06-27:
Ok, I believe I have already tried this and ran into issues (as the realsense camera feed into a realsense_splitter node, but I will try this again) 

Thanks