NVIDIA-ISAAC-ROS/isaac_ros_nvblox #112: Unable to run apt update, file has unexpected size

innowing-robotics-intern, 2024-06-17:
I’m trying to install nvblox on ubuntu22.04 on my Jetson agx orin. I successfully downloaded and launched the `isaac_ros_nvblox` package last week without updating the docker environment, so I have to rebuild the whole package again when I get into the docker environment. But when I tried to reinstall the nvblox following the steps in quick start, I got an error in `sudo apt update`, the following are the detailed error.
`$ sudo apt update 
Hit:1 https://packages.microsoft.com/repos/code stable InRelease
Hit:2 https://download.docker.com/linux/ubuntu jammy InRelease                 
Hit:3 https://repo.download.nvidia.com/jetson/common r36.3 InRelease           
Hit:4 https://repo.download.nvidia.com/jetson/t234 r36.3 InRelease             
Hit:5 https://repo.download.nvidia.com/jetson/ffmpeg r36.3 InRelease           
Hit:6 http://packages.ros.org/ros2/ubuntu jammy InRelease                      
Hit:7 http://ports.ubuntu.com/ubuntu-ports jammy InRelease                     
Hit:8 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease             
Get:9 https://pkgs.tailscale.com/stable/ubuntu jammy InRelease                 
Hit:10 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease          
Hit:11 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Hit:12 https://isaac.download.nvidia.cn/isaac-ros/release-3 jammy InRelease
Ign:13 https://isaac.download.nvidia.cn/isaac-ros/release-3 jammy/release-3.0 arm64 Packages
Get:13 https://isaac.download.nvidia.cn/isaac-ros/release-3 jammy/release-3.0 arm64 Packages [23.1 kB]
Err:13 https://isaac.download.nvidia.cn/isaac-ros/release-3 jammy/release-3.0 arm64 Packages
  File has unexpected size (23003 != 23063). Mirror sync in progress? [IP: 118.212.138.164 443]
  Hashes of expected file:
   - Filesize:23063 [weak]
   - SHA256:d1728a482b4422e7c4446155bd9d10a5ca39084967cbdd2e8a68979b0f463883
   - SHA1:9d487ab80af1da80a1192d3f100558bfca8c82e9 [weak]
   - MD5Sum:9f4542a47d445b350fc0bf470858103f [weak]
  Release file created at: Fri, 31 May 2024 17:11:20 +0000
Fetched 6,575 B in 5s (1,395 B/s)
Reading package lists... Done
W: https://isaac.download.nvidia.cn/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
E: Failed to fetch https://isaac.download.nvidia.cn/isaac-ros/release-3/dists/jammy/release-3.0/binary-arm64/Packages.bz2  File has unexpected size (23003 != 23063). Mirror sync in progress? [IP: 118.212.138.164 443]
   Hashes of expected file:
    - Filesize:23063 [weak]
    - SHA256:d1728a482b4422e7c4446155bd9d10a5ca39084967cbdd2e8a68979b0f463883
    - SHA1:9d487ab80af1da80a1192d3f100558bfca8c82e9 [weak]
    - MD5Sum:9f4542a47d445b350fc0bf470858103f [weak]
   Release file created at: Fri, 31 May 2024 17:11:20 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.`

nhathout, 2024-06-17:
I am getting the same problem. Trying to run apt-get upgrade and see if it makes a difference

alexmillane, 2024-06-24:
This issue should now be fixed.