NVIDIA-ISAAC-ROS/isaac_ros_nvblox #130: Issue while building from sourec isaac_ros_nvblox

adithyakumarcr, 2024-12-16:
Hi,

I am trying to build isaac_ros_nvblox from source. I am following the instructions from (https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_nvblox/isaac_ros_nvblox/index.html#quickstart) I find myself stuck with this issue for past couple of days. I tried different releases until 3.0.1, its still the same. The fact that surprises me the most is Until Nov 27th I didnt have issues with the build. I tried with multiple machines, still its the same. It would be great if someone can help me out. 

Thanks. 
```
Processing: nvblox_ros]                                        
--- stderr: nvblox_ros                                          
/workspaces/isaac_ros-dev/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp: In member function ‘void nvblox::NvbloxNode::subscribeToTopics()’:
/workspaces/isaac_ros-dev/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:179:38: error: ‘NitrosStatisticsConfig’ is not a member of ‘nvidia::isaac_ros::nitros’; did you mean ‘NitrosDiagnosticsConfig’?
  179 |           nvidia::isaac_ros::nitros::NitrosStatisticsConfig(), input_qos));
      |                                      ^~~~~~~~~~~~~~~~~~~~~~
      |                                      NitrosDiagnosticsConfig
/workspaces/isaac_ros-dev/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:195:38: error: ‘NitrosStatisticsConfig’ is not a member of ‘nvidia::isaac_ros::nitros’; did you mean ‘NitrosDiagnosticsConfig’?
  195 |           nvidia::isaac_ros::nitros::NitrosStatisticsConfig(), input_qos));
      |                                      ^~~~~~~~~~~~~~~~~~~~~~
      |                                      NitrosDiagnosticsConfig
gmake[2]: *** [CMakeFiles/nvblox_ros_lib.dir/build.make:230: CMakeFiles/nvblox_ros_lib.dir/src/lib/nvblox_node.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:535: CMakeFiles/nvblox_ros_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
```

alexmillane, 2024-12-23:
I just tried and am unable to repro this issue.

Could you try again with the latest release? You should have two repos checked out: `isaac_ros_common` and `isaac_ros_nvblox` both on the `release-3.2` branch. Under that condition, I was able to build.

jaiveersinghNV, 2025-06-17:
The link in the first post of this thread is evergreen and always points to the docs for the latest version of Isaac ROS, so the setup instructions listed there configure your apt sources to always fetch the latest `3.x` packages.

If you want to anchor to a specific release like 3.0 or 3.1, you will need to change the component to `legacy-release-3.x` in the source instructions:
https://nvidia-isaac-ros.github.io/v/release-3.1/getting_started/isaac_apt_repository.html#setup-source